cocotb/cocotb #669: Provide "DEBUG" Makefile variable

cmarqu, 2018-10-24:
For the kind of debugging work in #639, it would be good to have a documented Makefile variable like ``DEBUG`` or ``COCOTB_DEBUG`` which would switch options like Cadence Incisive/Xcelium's ``-plinowarn`` and ``-pliverbose``.
Does anyone know examples of such switches for other simulators?

Example code:
```
ifeq ($(DEBUG),1)
    EXTRA_ARGS += -pliverbose
    EXTRA_ARGS += -messages
else
    EXTRA_ARGS += -plinowarn
endif
```

cmarqu, 2018-10-28:
This is loosely related to #168.

cmarqu, 2020-03-08:
We should also merge ``VERILATOR_SIM_DEBUG`` and ``COCOTB_NVC_TRACE`` into this.

cmarqu, 2020-03-08:
The documentation should make clear that these are simulator options, not options for either cocotb's C/C++ code nor the log level for the logging module.

cmarqu, 2020-03-08:
An alternative would be to not have this variable at all but just add examples of e.g. additional ``EXTRA_ARGS`` to the documentation.

ktbarrett, 2020-03-09:
My opinion is if using these options is common, and beginners would benefit from using them, they should be separate variables and documented well. I think a "debug" variable that "does the right thing" for each simulator falls into that case.

cmarqu, 2020-03-09:
@wallento What do you think about the idea of throwing together ``VERILATOR_SIM_DEBUG`` and ``VERILATOR_TRACE`` under ``COCOTB_SIM_DEBUG``?

cmarqu, 2020-05-23:
Closing as we want to get rid of as many variables as possible, see #1495.

closed 2020-05-23 by cmarqu