cocotb/cocotb #678: Sort of linking of context for tracking moving in and out of python

stuarthodgson, 2018-10-28:
Ive merged #620 to unblock affected people but would like to fix it properly too.

``move the to_ functions into GPI so there won't be a need to link with cocotbutils.```

ktbarrett, 2020-05-06:
I proper fix would be to separate the cocotb_utils into two libraries: cocotb_utils and gpi_utils.

cocotb_utils will be linked by gpi_embed and the simulator module, while gpi_utils would be linked by the GPI, potentially the simulator modules, and gpi_embed and simulator module. cocotb_utils would contain the `to_*` methods only (they are only used by gpi_embed and the simulator module), while the other methods would be used by the other libraries.

eric-wieser, 2020-05-06:
> cocotb_utils would contain the `to_*` methods only

These are only macros through, so cannot be put in a library?

ktbarrett, 2020-05-06:
That's a nomenclature thing, a library can be header only.