cocotb/cocotb #679: Error using cocotb with Modelsim DE

tmeissner, 2018-10-29:
In lines 30-31 of Makefile.modelsim it is assumed that Modelsim is a 64 bit binary.

```
# Modelsim is (no longer) 32-bit only
#ARCH:=i686
```

Maybe there are versions of Modelsim that are 64 bit, but Modelsim DE isn't. So the build fails when trying to use the vish binary under `modelsim_dlx/linux_x86_64pe/` folder:

```
make[2]: Leaving directory `/home/meissner/projects/github/cocotb.git/lib/fli'
set -o pipefail; cd sim_build && LD_LIBRARY_PATH=/home/meissner/projects/github/cocotb.git/build/libs/x86_64::/usr/lib:/usr/lib:/usr/lib:/usr/lib MODULE=test_adder TESTCASE= TOPLEVEL=adder \
        GPI_EXTRA= TOPLEVEL_LANG=verilog PYTHONPATH=/home/meissner/projects/github/cocotb.git/build/libs/x86_64:/home/meissner/projects/github/cocotb.git:/home/meissner/projects/github/cocotb.git/examples/adder/tests:/home/meissner/projects/github/cocotb.git/examples/adder/tests/../model: \
        /opt/mentor/modelsim_de_10.7b/modelsim_dlx/linuxpe/vsim -c -64 -do runsim.do 2>&1 | tee sim.log
Failed to open executable /opt/mentor/modelsim_de_10.7b/modelsim_dlx/linuxpe/../linux_x86_64pe/vish in execute mode needed for the option -64.
execv: No such file or directory
** Fatal: Unable to exec the GUI /opt/mentor/modelsim_de_10.7b/modelsim_dlx/linuxpe/../linux_x86_64pe/vish.
make[1]: *** [results.xml] Error 1
make[1]: Leaving directory `/home/meissner/projects/github/cocotb.git/examples/adder/tests'
make: *** [sim] Error 2
```

I think, this line has to be removed for the case of Modelsim DE (which is Linux only, if I remember correctly). Or a case differentiation has to be done dependent on the specific Modelsim version (PE or DE).

closed 2018-10-29 by tmeissner

mrkrcc, 2023-05-17:
any solution ?
is it possible to work with modelsim(vhdl) on 64bit ubuntu?

`home/mrkrc/intelFPGA/18.1/modelsim_ase/bin/vsim -c -64  -do sim_build/runsim.do  2>&1 | tee sim_build/sim.log
Failed to open executable /home/mrkrc/intelFPGA/18.1/modelsim_ase/bin/../linux/../linux_x86_64pe/vish in execute mode needed for the option -64.
execv: No such file or directory`

Botnic, 2023-05-19:
Yes should be possible. I use Modelsim 32bit on a RHL8 64bit.
You need a 32bit version of Python3. Then set the parameter ARCH=i686.

themperek, 2023-05-23:
https://github.com/cocotb/cocotb/wiki/Tier-2-Setup-Instructions#redhat-based-installation

mrkrcc, 2023-05-25:
thank you thomasz. main problem is 32-bit python installation. everything works fine for now.