cocotb/cocotb #690: simulator exits early if first nested coroutine exits without yielding trigger

s3bs, 2018-11-14:
Sample code snippet:
```
@cocotb.test()
def test_premature_exit(dut):
    @cocotb.coroutine
    def set_valid(signal, wait=False):
        if wait:
            yield Timer(10, "ns")
        signal <= 1
    @cocotb.coroutine
    def enable_tb(dut):
        yield set_valid(dut.stream_in_valid)
        yield Timer(10, "ns")

    yield enable_tb(dut)
```
leads to (run as part of test_cocotb.py at the very end)
```
# COUT:  2533.28ns ERROR    cocotb.scheduler                            __init__.py:177  in _sim_event                      Failing test at simulator request before test run completion: Simulator shutdown prematurely
# COUT:  2533.28ns DEBUG    cocotb.scheduler                           scheduler.py:698  in finish_scheduler                Issue sim closedown result to regresssion object
# COUT:  2533.28ns ERROR    cocotb.regression                         regression.py:290  in handle_result                   Test error has lead to simulator shuttting us down
```
With my understanding of the scheduler, this is due to the fact that we never entered `react` at some point before we return from the `set_valid` coroutine. During unschedule the `_join` trigger is only appended. The problem in this case is that that trigger won't ever be executed. 

Even as little as yielding a `NullTrigger` inside of `set_valid` would not cause this issue
Here the full trace with `COCOTB_SCHEDULER_DEBUG=1 COCOTB_LOG_LEVEL=DEBUG SIM=aldec`

```
# COUT:  2533.28ns INFO     cocotb.regression                         regression.py:318  in execute                         Running test 22/27: test_premature_exit
# COUT:  2533.28ns DEBUG    ..est_premature_exit.0x7fffbb0ac750        scheduler.py:594  in schedule                        Scheduling with None
# COUT:  2533.28ns INFO     ..est_premature_exit.0x7fffbb0ac750       decorators.py:211  in send                            Starting test: "test_premature_exit"
# COUT:                                                                                                                     Description: None
# COUT:  2533.28ns DEBUG    ..est_premature_exit.0x7fffbb0ac750       decorators.py:219  in send                            Sending trigger None
# COUT:  2533.28ns DEBUG    cocotb.scheduler                           scheduler.py:600  in schedule                        Coroutine test_premature_exit yielded enable_tb (mode 1)
# COUT:  2533.28ns DEBUG    cocotb.scheduler                           scheduler.py:628  in schedule                        Scheduling nested co-routine: enable_tb
# COUT:  2533.28ns DEBUG    cocotb.scheduler                           scheduler.py:562  in add                             Adding new coroutine enable_tb
# COUT:  2533.28ns DEBUG    ..oroutine.enable_tb.0x7fffb9771950        scheduler.py:594  in schedule                        Scheduling with None
# COUT:  2533.28ns DEBUG    cocotb.gpi                                GpiCommon.cpp:256  in __gpi_get_handle_by_name        Searching for stream_in_valid
# COUT:  2533.28ns DEBUG    cocotb.gpi                                GpiCommon.cpp:269  in __gpi_get_handle_by_name        Checking if stream_in_valid native though impl VPI
# COUT:  2533.28ns DEBUG    cocotb.gpi                                 VpiCbHdl.cpp:279  in initialise                      VPI: stream_in_valid initialised with 1 elements
# COUT:  2533.28ns DEBUG    cocotb.gpi                                  VpiImpl.cpp:208  in create_gpi_obj_from_handle      VPI: Created object with type was vpiNet(36)
# COUT:  2533.28ns DEBUG    cocotb.gpi                                GpiCommon.cpp:279  in __gpi_get_handle_by_name        Found stream_in_valid via VPI
# COUT:  2533.28ns DEBUG    cocotb.gpi                                GpiCommon.cpp:50   in check_and_store                 Checking sample_module.stream_in_valid exists
# COUT:  2533.28ns DEBUG    cocotb.stream_in_valid                        handle.py:92   in __init__                        Created
# COUT:  2533.28ns DEBUG    cocotb.scheduler                           scheduler.py:600  in schedule                        Coroutine enable_tb yielded set_valid (mode 1)
# COUT:  2533.28ns DEBUG    cocotb.scheduler                           scheduler.py:628  in schedule                        Scheduling nested co-routine: set_valid
# COUT:  2533.28ns DEBUG    cocotb.scheduler                           scheduler.py:562  in add                             Adding new coroutine set_valid
# COUT:  2533.28ns DEBUG    ..oroutine.set_valid.0x7fffb9792050        scheduler.py:594  in schedule                        Scheduling with None
# COUT:  2533.28ns DEBUG    cocotb.scheduler                           scheduler.py:613  in schedule                        Coroutine completed: set_valid
# COUT:  2533.28ns DEBUG    cocotb.gpi                                 GpiCbHdl.cpp:128  in run_callback                    Generic run_callback done
# COUT:  2533.28ns DEBUG    cocotb.gpi                                 GpiCbHdl.cpp:126  in run_callback                    Generic run_callback
# COUT:  2533.28ns DEBUG    cocotb.scheduler                           scheduler.py:314  in react                           Trigger fired: _ReadWrite(readwritesync)
# COUT:  2533.28ns DEBUG    cocotb.scheduler                           scheduler.py:334  in react                           Writing cached signal updates
# COUT:  2533.28ns DEBUG    cocotb.gpi                                   gpi_priv.h:129  in get_num_elems                   stream_in_valid has 1 elements
# COUT:  2533.28ns DEBUG    cocotb.gpi                                 GpiCbHdl.cpp:128  in run_callback                    Generic run_callback done
# KERNEL: Simulation has finished. There are no more test vectors to simulate.
endsim
# COUT:  2533.28ns ERROR    cocotb.scheduler                            __init__.py:177  in _sim_event                      Failing test at simulator request before test run completion: Simulator shutdown prematurely
# COUT:  2533.28ns DEBUG    cocotb.scheduler                           scheduler.py:698  in finish_scheduler                Issue sim closedown result to regresssion object
# COUT:  2533.28ns ERROR    cocotb.regression                         regression.py:289  in handle_result                   Test error has lead to simulator shutting us down
# COUT:  2533.28ns ERROR    cocotb.regression                         regression.py:207  in tear_down                       Failed 1 out of 22 tests (3 skipped)
```

eric-wieser, 2019-03-17:
I think this is a dupe of the now-fixed #637, but might be worth verifying

eric-wieser, 2019-03-17:
Nope, still broken

eric-wieser, 2019-03-17:
From the test added in #637:
```
    # note: it seems that the test still has to yield at least once, even
    # if the subroutines do not
    yield Timer(1)
```

ktbarrett, 2019-10-01:
This can be closed. Just tested and the code snippet works.

eric-wieser, 2019-10-01:
Note that #637 still has issues with external (which I've made a start on fixing), but this specific case is indeed fixed.

closed 2020-07-10 by ktbarrett