cocotb/cocotb #696: python control file/function (instead of Makefile)

themperek, 2018-11-15:
What would be your opinion to have python control file for the cocotb instead of makefile.

Initially one can start with simple a wrapper for Makefile. A module that will generate a local Makefile based on python code and execute it (subprocess). Later can recede makefiles to just compile liberalises and run simulator directly form python.

Could look something like this at the end of test file:
```python
from cocotb.runcontrol import Run

def test_run():
    Run(
        sources = ['../hdl/dff.v'],
        top_level = 'dff',
        incdir = ['../hdl'],
        defines=['TEST=1'],
    )

if __name__ == "__main__":
    test_run()
```

This could automatic discoverable by **pytest**. If we get #517 we would run in parallel.

It need bit of thinking and trying but would like to know your opinion.
Somehow related to #437, #508, #667
In case we positive feedback can make a working demo.

cmarqu, 2018-11-15:
I like the idea.
Starting to use cocotb on an existing design with testbench infrastructure is not easy, this would make it easier.

tmeissner, 2018-11-16:
The `cocotb-testrunner` looks pretty cool. The yaml manifest is easy to read, which is good. I like the idea.

imphil, 2018-11-16:
> `cocotb-testrunner` is VCS-only right now, correct?

It's VCS-only since it only writes (cocotb standard) Makefiles for VCS. It shouldn't be hard to write other Makefiles as well, I just did never try it (as I don't have use for it).

imphil, 2018-11-16:
> > Add a yaml "manifest" file describing your design to a test directory.
> 
> I like yaml but any advantage of yaml over having it part of python test file? (Potentially much more control is possible?).

I think both works, it's more a matter of taste. In the end I don't think it highly matters.

In the longer run I'm planning to use fusesoc (which is the "parent" of edalize @cmarqu mentioned) to generate the test description (and in the end abandon the manifest file altogether). All relevant information is already in the fusesoc *.core files (e.g. https://github.com/opensocdebug/osd-hw/blob/osd-next/modules/stm/stm_mor1kx.core), especially the list of source files with all dependencies.

About running in parallel: cocotb-testrunner already supports it, but reusing pytest for that would probably be a really good idea.

cmarqu, 2018-11-16:
Coming again from the use case where I have an existing non-cocotb testbench setup that I want to keep using, getting the list of files from Python instead of from a YAML file makes it possible to query my existing setup.
Querying could mean getting the file list (or multiple versions of it, like one for an ASIC, one for an FPGA prototype) from IP-XACT, or parsing ``*.f`` (Cadence) or ``*.do`` (Mentor) files.
Of course, the querying tool could dump the YAML file as an intermediate, but ... layers all the way down! :)

themperek, 2018-11-16:
>I think both works, it's more a matter of taste. In the end I don't think it highly matters.

Actually one should be able to do something like this:
```python
from cocotb.runcontrol import Run

def test_run(): 
    Run(**yaml.load('manifest.yaml'))

if __name__ == "__main__":
    test_run()
```

themperek, 2020-11-17:
The initial version is available here: https://github.com/themperek/cocotb-test/ can be integrated into cocotb.

closed 2020-11-17 by themperek