cocotb/cocotb #700: SystemError from Python - race condition at startup?

cmarqu, 2018-11-18:
Hi,

I'm trying to simulate a rather large design with cocotb in Cadence Xcelium, and in maybe 70% of all runs, I'getting the following error:
```
     -.--ns INFO     cocotb.gpi                                  gpi_embed.c:111  in embed_init_python               Did not detect Python virtual environment. Using system-wide Python interpreter.
     -.--ns INFO     cocotb.gpi                                GpiCommon.cpp:92   in gpi_print_registered_impl       VPI registered
SystemError: Objects/listobject.c:307: bad argument to internal function

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/cmarqu/tools/cocotb/cocotb/log.py", line 113, in _willLog
    return self.logger.isEnabledFor(level)
  File "/tools/anaconda/4.3.0-python3.6/Linux-x86_64/lib/python3.6/logging/__init__.py", line 1543, in isEnabledFor
    return level >= self.getEffectiveLevel()
SystemError: PyEval_EvalFrameEx returned a result with an error set

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/cmarqu/tools/cocotb/cocotb/log.py", line 113, in _willLog
    return self.logger.isEnabledFor(level)
  File "/tools/anaconda/4.3.0-python3.6/Linux-x86_64/lib/python3.6/logging/__init__.py", line 1543, in isEnabledFor
    return level >= self.getEffectiveLevel()
SystemError: PyEval_EvalFrameEx returned a result with an error set

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/cmarqu/tools/cocotb/cocotb/__init__.py", line 103, in _initialise_testbench
    _rlock.acquire()
SystemError: <built-in method acquire of _thread.RLock object at 0x7fa4bf1b6540> returned a result with an error set
Cocotb initialisation failed - exiting
    0.00ns ERROR    cocotb.scheduler                            __init__.py:177  in _sim_event                      Failing test at simulator request before test run completion: Simulator shutdown prematurely
```
The other 30% of all runs, everything goes smoothly. Any ideas?

mciepluc, 2018-12-14:
@cmarqu which version are you using? I am just running on 18.03.012 and not observing such issue.

cmarqu, 2018-12-14:
This was with 18.09. 
Right now, and several compilations later, I don't see this problem anymore, so it may have been some glitch in whatever. I'm closing this for now.

closed 2018-12-14 by cmarqu