cocotb/cocotb #711: PLUSARGS handling with QuestaSim

mciepluc, 2018-12-06:
Hi,

I noted that ```PLUSARGS``` is not passed to the Python correctly under Questasim (and maybe other simulators as well). It seems to work as expected under Icarus. Considering the test [test_plusargs](https://github.com/potentialventures/cocotb/blob/master/tests/test_cases/test_plusargs/plusargs.py) the results is OK in Icarus:
```
/usr/local/bin/vvp -M /home/mci/cocotb-pv/build/libs/x86_64 -m gpivpi sim_build/sim.vvp   +foo=bar +test1 +test2 +options=fubar
     -.--ns INFO     cocotb.gpi                                  gpi_embed.c:111  in embed_init_python               Did not detect Python virtual environment. Using system-wide Python interpreter.
     -.--ns INFO     cocotb.gpi                                GpiCommon.cpp:91   in gpi_print_registered_impl       VPI registered
    0.00ns INFO     cocotb.gpi                                  gpi_embed.c:278  in embed_sim_init                  Running on Icarus Verilog version 0.10.0 (devel)
    0.00ns INFO     cocotb.gpi                                  gpi_embed.c:279  in embed_sim_init                  Python interpreter initialised and cocotb loaded!
    0.00ns INFO     cocotb                                      __init__.py:118  in _initialise_testbench           Running tests with Cocotb v1.0.1 from /home/mci/cocotb-pv
    0.00ns INFO     cocotb                                      __init__.py:134  in _initialise_testbench           Seeding Python random module with 1544089045
    0.00ns INFO     cocotb.regression                         regression.py:187  in _initialise                     Found test plusargs.plusargs_test
    0.00ns INFO     cocotb.regression                         regression.py:319  in execute                         Running test 1/1: plusargs_test
    0.00ns INFO     ..tine.plusargs_test.0x7f43a6bccf60       decorators.py:211  in send                            Starting test: "plusargs_test"
                                                                                                                    Description: Demonstrates plusarg access from Python test
SIM: Plusargs test
SIM: Plusarg foo has value bar
SIM: Test running
COCOTB: foo bar
COCOTB: test1 True
COCOTB: test2 True
COCOTB: options fubar
20001000000000.00ns INFO     cocotb.regression                         regression.py:264  in handle_result                   Test Passed: plusargs_test
```

The same test run under Questa:

```
/opt/tools/ams/amsv_2017_3_1/amsv/modeltech/bin/vsim -c -64 -do runsim.do 2>&1 +foo=bar +test1 +test2 +options=fubar | tee sim.log
...
Reading pref.tcl
# Loading sv_std.std
# Loading work.tb_top
# Loading /home/mci/cocotb-pv/build/libs/x86_64/libvpi.so
#      -.--ns INFO     cocotb.gpi                                  gpi_embed.c:111  in embed_init_python               Did not detect Python virtual environment. Using system-wide Python interpreter.
#      -.--ns INFO     cocotb.gpi                                GpiCommon.cpp:91   in gpi_print_registered_impl       VPI registered
#     0.00ns INFO     cocotb.gpi                                  gpi_embed.c:278  in embed_sim_init                  Running on ModelSim for Questa-64 version 10.6c_1 2017.09
#     0.00ns INFO     cocotb.gpi                                  gpi_embed.c:279  in embed_sim_init                  Python interpreter initialised and cocotb loaded!
#     0.00ns INFO     cocotb                                      __init__.py:118  in _initialise_testbench           Running tests with Cocotb v1.0.1 from /home/mci/cocotb-pv
#     0.00ns INFO     cocotb                                      __init__.py:134  in _initialise_testbench           Seeding Python random module with 1544090493
#     0.00ns INFO     cocotb.regression                         regression.py:187  in _initialise                     Found test plusargs.plusargs_test
#     0.00ns INFO     cocotb.regression                         regression.py:319  in execute                         Running test 1/1: plusargs_test
#     0.00ns INFO     ..tine.plusargs_test.0x7fffe9ffe470       decorators.py:211  in send                            Starting test: "plusargs_test"
#                                                                                                                     Description: Demonstrates plusarg access from Python test
# SIM: Plusargs test
# SIM: Plusarg foo has value 
# SIM: Test running
#  2000.10ns INFO     cocotb.regression                         regression.py:264  in handle_result                   Test Passed: plusargs_test

```

Interestingly, arguments are passed correctly under Questa by ```EXTRA_ARGS``` variable. 
Any clues what may be going on?

themperek, 2020-03-25:
@ktbarrett This can be closed?

ktbarrett, 2020-03-25:
yep

themperek, 2020-03-25:
@mciepluc It should work now. In case you have still problems please reopen,

closed 2020-03-25 by themperek