cocotb/cocotb #712: Setting Scoreboard fail_immediately=False but strict_type=True causes all tests to pass.

kolojoe, 2018-12-07:
If Scoreboard is created with `fail_immediately=False` but an interface or comparison function is created with `strict_type=True` all tests will pass as long as a transaction arrives.

kolojoe, 2018-12-07:
Sure, I'll try to come up with something.

Paraphrasing a bit of code here...

Scoreboard
```python
self.expected = []
self.scoreboard = Scoreboard(self.dut, fail_immediately=False)
self.scoreboard.add_interface(self.clockedmonout, self.expected,
                              strict_type=True)
```

Model within input monitor.
```python
def model(self, transaction):
    out = []
    self.expected.append(out)
```

Output monitor receive
```python
@cocotb.coroutine
def _monitr_recv(self):
    while True:
        vals = {}
        self._recv(vals)
```

The output showing erroneous passing of tests.  Pardon the formatting, just scroll to the right.
```
11:20:29.470 | INFO       | ...                      | ...              | #       252.00ns | ERROR      | Received transaction is a different type to expected transaction
11:20:29.470 | INFO       | ...                      | ...              | #       252.00ns | INFO       | Got: <class 'dict'> but expected <class 'list'>
11:20:29.470 | INFO       | ...                      | ...              | #       258.00ns | ERROR      | Received transaction is a different type to expected transaction
11:20:29.731 | INFO       | ...                      | ...              | #       258.00ns | INFO       | Got: <class 'dict'> but expected <class 'list'>
             |            |                          |                  | #       263.00ns | INFO       | Test Passed: testreg
             |            |                          |                  | #       263.00ns | INFO       | Passed 1 tests (0 skipped)
             |            |                          |                  | #       263.00ns | INFO       | **********************************************************************
             |            |                          |                  | #                |            | ** TEST          PASS/FAIL  SIM TIME(NS)  REAL TIME(S)  RATIO(NS/S) **
             |            |                          |                  | #                |            | **********************************************************************
             |            |                          |                  | #                |            | ** test.testreg    PASS          263.00          0.18      1476.99  **
             |            |                          |                  | #                |            | **********************************************************************
             |            |                          |                  | #                |            |
             |            |                          |                  | #       263.00ns | INFO       | *************************************************************************************
             |            |                          |                  | #                |            | **                                 ERRORS : 0                                      **
             |            |                          |                  | #                |            | *************************************************************************************
             |            |                          |                  | #                |            | **                               SIM TIME : 263.00 NS                              **
             |            |                          |                  | #                |            | **                              REAL TIME : 0.21 S                                 **
             |            |                          |                  | #                |            | **                        SIM / REAL TIME : 1230.17 NS/S                           **
             |            |                          |                  | #                |            | *************************************************************************************
             |            |                          |                  | #                |            |
             |            |                          |                  | #       263.00ns | INFO       | Shutting down...
             |            |                          |                  | # Simulation halt requested by foreign interface.
```

cmarqu, 2018-12-07:
Not that I mind scrolling, but ``COCOTB_REDUCED_LOG_FMT`` helps. :)

kolojoe, 2019-01-07:
@imphil @cmarqu 
Upon further examination it seems that this is accounted for, but the code that is supposed to cause a failure isn't being called.  The solution that is implemented is a simple `errors` counter that should cause the TestFailure exception to be raised, but it seems to not ever be called... Could anyone give some immediate insight into what the `result` property is for/if it's being used?

Here is the code that I'm referring to.

`cocotb/scoreboard.py`
```python
class Scoreboard(object):
.
.
.
    @property
    def result(self):
        """Determine the test result, do we have any pending data remaining?"""
        fail = False
        for monitor, expected_output in self.expected.items():
            if callable(expected_output):
                self.log.debug("Can't check all data returned for %s since "
                               "expected output is callable function rather "
                               "than a list" % str(monitor))
                continue
            if len(expected_output):
                self.log.warn("Still expecting %d transactions on %s" %
                              (len(expected_output), str(monitor)))
                for index, transaction in enumerate(expected_output):
                    self.log.info("Expecting %d:\n%s" %
                                  (index, hexdump(str(transaction))))
                    if index > 5:
                        self.log.info("... and %d more to come" %
                                      (len(expected_output) - index - 1)) 
                        break
                fail = True
        if fail:
            return TestFailure("Not all expected output was received")
        if self.errors:
            return TestFailure("Errors were recorded during the test")
        return TestSuccess()

    def compare(self, got, exp, log, strict_type=True):
        """ 
        Common function for comparing two transactions.

        Can be re-implemented by a subclass.
        """

        # Compare the types
        if strict_type and type(got) != type(exp):
            self.errors += 1
            log.error("Received transaction type is different than expected")
            log.info("Received: %s but expected %s" %
                     (str(type(got)), str(type(exp))))
            if self._imm:
                raise TestFailure("Received transaction of wrong type. "
                                  "Set strict_type=False to avoid this.")
            return
.
.
.
```

cmarqu, 2019-01-09:
As a small data point, it appears to be used here: 
https://github.com/potentialventures/cocotb/search?q=%22scoreboard.result%22&unscoped_q=%22scoreboard.result%22