cocotb/cocotb #715: Change Copyright message in modules from docstring to single line comments.

kolojoe, 2018-12-07:
When using the `automodule` directive in the rst files for documentation the large copyright docstring will be inserted a documentation for the module.  If we change this to being inline comments then we can have an actual docstring for modules and subsequently autodocument them using sphinx instead of calling out each class.

cmarqu, 2018-12-07:
I agree. PEP 257 says
> The docstring for a module should generally list the classes, exceptions and functions (and any other objects) that are exported by the module, with a one-line summary of each.

so implicitly disallows that IMHO.

cmarqu, 2019-01-13:
I guess we could start using ``automodule`` now and convert the modules not already handled by the above PR as we go.