cocotb/cocotb #718: Specify path to libpython<ver>.so at build time

TrampHRC, 2018-12-10:
I am using cocotb with modelsim 10.6c and python3.6, not through the standard makefiles but through my service's custom TCL scripts. Actually I made a makefile.compile, to build just cocotb and then i use the tcl script to run simulation.
I got the following error while loading libfli.so :
`error while loading shared libraries: libpython3.6m.so.1.0: cannot open shared object file: No such file or directory`

I now know that setting LD_LIBRARY_PATH to the directory of libpython3.6m.so.1.0 solves the problem. It is done that way, it seems, with the makefiles.
Though after reading through this very good [blog](https://amir.rachum.com/blog/2016/09/17/shared-libraries/) I got that you could also solve this issue at build time with the right command on the right file creation. 
I feel like this would be a better solution because then the information is inside the .so and you can remove it from scripts that would focus solely on simulation.

Reading a bit through the lib/Makefile, I don't really feel like this can be done through env variables, so I would have to change the Makefiles.
Using ldd on the whole build directory, I can get a list of the files that depend on my libpython3.6m.so.1.0.
When building these files, I should add
`-Wl,-rpath,/path/to/python/lib`
But I'm not sure how, as I am not a Makefile expert...

Is there a reason why this was done through setting LD_LIBRARY_PATH to PYTHON_LIBDIR in makefiles ?
Could I get any advice on how to change my Makefiles ?

Thank you

cmarqu, 2018-12-10:
I believe the Makefile part you need to modify starts at (specific to the operating system) https://github.com/potentialventures/cocotb/blob/e86924503bbb02a78f37c41716f9af56fb5fa6a1/makefiles/Makefile.inc#L89
As for your other question, the authors will have to chime in.

TrampHRC, 2018-12-12:
That's it ! Thank you @cmarqu 
It looks like this for my Linux OS:
`else ifeq ($(OS),Linux)
SO_LINKER_ARGS := -shared -Xlinker -L$(PYTHON_LIBDIR) -Wl,--enable-new-dtags,-rpath,$(PYTHON_LIBDIR)`
and it cuts the need to put PYTHON_LIBDIR in my LD_LIBRARY_PATH.
One could also simply put
`-Wl,-rpath,$(PYTHON_LIBDIR)`
And I let you read the blog i referenced above to see what this difference implies, but in our usecase it's pretty equivalent.

closed 2018-12-12 by TrampHRC

cmarqu, 2018-12-13:
You may also be interested in #696 once it materialises.