cocotb/cocotb #719: Switch from `property()` to `@property` syntax

imphil, 2018-12-12:
Since Python 2.6 the @property/@x.setter syntax is supported, and gives a cleaner syntax than `property()`. Convert existing uses of `property()` to the newer syntax. 

This requires figuring out what to do with the existing setters and getter functions, which are often public (i.e. set_x() and get_x()). If existing code depends on these, we probably should keep compat wrappers and issue a deprecation warning.

eric-wieser, 2018-12-15:
> we probably should keep compat wrappers

An easy trick for this is
```python
@property
def my_prop(self):
    ...

@my_prop.setter
def my_prop(self, value):
    ...

# aliases for compatibility
set_my_prop = my_prop.fset
get_my_prop = my_prop.fset
```

eric-wieser, 2020-03-19:
Ah, not addressed by gh-1497: Deprecating these. I'll open a new issue.

commit 8093321c (2020-03-19): Create BinaryValue properties using decorators. (#1497)

This expresses functions like `get_value` and `set_value` in terms of the property `value`, rather than the other way around.
These accessor functions are no longer included in the docs, but will continue to work.

The behavior of the properties themselves is unchanged.

Additionally, the supposedly-deprecated `.value` property is now expressed in terms of the non-deprecated `.integer` property, rather than vice-versa.

Closes #719.
closed 2020-03-19 by eric-wieser