cocotb/cocotb #724: Release/tag cocotb before

imphil, 2018-12-15:
We're about to finish the work on the packaging of cocotb in #517. Users updating cocotb by pulling the repo will additionally need to adjust their COCOTB environment variable (or install the package).

Let's have a "release" (assign a version number/tag mostly) before we merge #517.

I'm suggesting to use version 1.1 (doesn't matter much for now, we need to figure out what compat guarantees we give between at a later point). 

What do you guys think? 

(Thanks Tomasz for the idea and the packaging work!)

mciepluc, 2018-12-15:
I was not really following package discussion, but it seems to be quite a major update. From my point of view yes, definitely let's release 1.1 before #517 merge. Before that, also I think it makes sense to do issues/other PRs classification, do we target to address them in 1.1 or afterwards.

cmarqu, 2018-12-15:
I'm also for a release that just fixes bugs and adds small features accumulated over the last years.
After that's done, we could (re-)start using milestones (https://github.com/potentialventures/cocotb/milestones) and should also have proper release notes.

imphil, 2018-12-15:
So my proposed plan would be:

- Release *now* a 1.1 release with whatever we have today. This will be the last release before the directory structure changes in #517. (You can always argue "take this one more PR", and then we never release.)
- Then merge #517, which changes the directory structure and adds the ability to build a pip package.
- Merge more outstanding work and let people test the package for a bit. Ideally provide a preview package on pypi to also test our deployment infrastructure/pipeline.
- Release a 1.2 version in January.

This should get us back on track with the most pressing items from the backlog cleared, and we can then discuss how to move on (e.g. regarding larger changes).

imphil, 2018-12-15:
@stuarthodgson @chiggs do you have other ideas/suggestions/vetoes on that?

stuarthodgson, 2018-12-15:
I would like to get the chiggs PR merged before #517 as that would allow us to use mainline rather than a fork internally and thus make us contribute more. But that pretty much means a slight delay in 1.1. 

The changes in #517 are not good for us in the short term. We can't just change COCOTB environment variables in one place. We would need to move to the package install (good!). So for our needs can we delay 1.1 until Monday.

themperek, 2018-12-15:
At the moment #517 should be backward compatible and we could keep it like this for some time (1.3?) though it is bit ugly.

stuarthodgson, 2018-12-15:
I'd missed the symlinks, for a version at least probably have to keep backwards compatible.

imphil, 2018-12-16:
@stuarthodgson sounds like a plan! Let's have #708 and a bit of time afterwards for you to see if you can go upstream, and then cut the release which works for you internally as well. That should of course be a priority.

mciepluc, 2018-12-16:
Guys, I agree with Stuart, there are also some other important PRs pending like #493, #601, #640 which will be ready to be merged soon. Maybe just plan the 1.1 release deadline and see how much we can do. I personally don't like the "release what we have" approach, as release should be considered pretty much stable. Then change from 1.1. to 1.2 could be ONLY merging #517. Just my opinion.

themperek, 2018-12-17:
Is there anything outside https://github.com/potentialventures/cocotb/pull/708 that you think MUST be in 1.1.0 ?

PS. https://en.wikipedia.org/wiki/Release_early,_release_often

themperek, 2019-01-06:
What do you think about releasing 1.1 now (#708 is in)? Maybe should add #734 before?

Before 1.2 one could marge #517 (package support) and wait a bit so we all check potential problem and compatibilities. 

For 1.2 I would also suggest:
  - add windows and osx to travis-ci
  - automatic upload to pipy on release 
  - update documentation on how to install (just before release)
  - whatever else gets resolved

imphil, 2019-01-06:
> What do you think about releasing 1.1 now (#708 is in)? Maybe should add #734 before?

Yes, the fix for the #729 is something we need for the release. But other than that there's currently nothing on the radar that's essential. Given that we just merged a huge number of commits let's give it a week or two to settle and then do a release. 

@stuarthodgson / @chiggs can you check that the current master works for you to switch back to the upstream code base? (Possibly with #734 applied for now.)

stuarthodgson, 2019-01-06:
Yep, will do now things are merged.

mciepluc, 2019-01-07:
I cannot test current master as long as #729 is present.

imphil, 2019-01-21:
All dependencies of the 1.1 milestone are resolved and nothing serious has come up over the last two weeks, I'm going to release 1.1 tomorrow to unblock merging more work that's already lining up. If your favorite PR didn't make it to this release, don't worry: the next release isn't far away.

imphil, 2019-01-23:
v1.1 has been released now, find the release announcement at https://lists.librecores.org/pipermail/cocotb/2019-January/000053.html

imphil, 2019-01-23:
The flood gates are open again for the next PRs, I'll try to have a bit of time for those in the coming days -- but please bear with me if not everything is merged instantly!

closed 2019-01-23 by imphil