cocotb/cocotb #725: Use tox to run tests in CI

imphil, 2018-12-15:
Currently we have a custom setup to run the tests in Travis CI on Python 2.7 and 3.5. As we run the tests from a development build of the pip package instead of the "real" one, this can lead to inconsistencies and adds complexity.

Tox should be able to solve this nicely and catch more packaging bugs early. Let's give it a try.

At the same time we can probably remove docker from the Travis testing and use plain Travis since the only thing we need to do is build and install Icarus Verilog, which should work equally well without docker.  This removes even more complexity and gives us the ability to use standard matrix builds with preinstalled Python versions (see https://docs.travis-ci.com/user/languages/python/#specifying-python-versions).

imphil, 2018-12-15:
@themperek I think you started on this one already?

themperek, 2018-12-15:
@imphil It is part of https://github.com/potentialventures/cocotb/pull/517 now. 
For the moment in docker. Mind we can add Windows soon. Running everything in docker may be cleaner.

imphil, 2018-12-15:
Travis can do native Windows since a couple of weeks, so that shouldn't hold us back.

themperek, 2018-12-15:
BTW: `osx` seems to work out of box: https://travis-ci.org/themperek/cocotb/builds/468525027

forensicgarlic, 2018-12-16:
personally I'm a fan of open source projects having a docker environment because it makes "try before you buy" a lot easier. It sounds like it's not buying cocotb a lot on the travis CI, but providing one can make setup as easy as copying a docker file. A docker file provided that's not used / tested is going to get bitrot though.

themperek, 2019-03-20:
Seems to be done.

closed 2019-03-20 by themperek