cocotb/cocotb #728: "undefined symbol: _Py_TrueStruct" on Python 3.6+

imphil, 2018-12-16:
cocotb is broken is Python 3.6 and 3.7:

~~~
vvp -M /home/travis/build/potentialventures/cocotb/build/libs/x86_64 -m gpivpi sim_build/sim.vvp   
gpivpi:`/home/travis/build/potentialventures/cocotb/build/libs/x86_64/gpivpi.vpl' failed to open using dlopen() because:
    /home/travis/build/potentialventures/cocotb/build/libs/x86_64/libgpilog.so: undefined symbol: _Py_TrueStruct.
~~~

A full log is available at https://travis-ci.org/potentialventures/cocotb/jobs/468756352 (that's test for a new CI configuration, so don't worry about anything else in this run failing)

mciepluc, 2018-12-17:
Isn't it the problem?
```
1500 pyenv: python3.6-config: command not found
1501
1502 The `python3.6-config' command exists in these Python versions:
1503  3.6
1504  3.6.7
```

imphil, 2018-12-17:
Still looking into it, the Makefiles have a lot of duct tape in various places, some of  which is essential, some of which seems to work around bugs somewhere else.

eric-wieser, 2020-04-09:
@imphil: can this be closed?

imphil, 2020-04-09:
I'm pretty sure it can. Thanks.

closed 2020-04-09 by imphil