cocotb/cocotb #729: A lot of "Attempt to prime an already primed trigger for cbValueChange!" messages

mciepluc, 2018-12-19:
Seems one of the latest merges caused additional production of a lot of such log messages:
```
Attempt to prime an already primed trigger for cbValueChange!
We seem to already be registered, deregistering cbValueChange!
```
I don't quite have time to look into it now, maybe someone know what is going on?

themperek, 2018-12-19:
@eric-wieser

eric-wieser, 2018-12-19:
It sounds like the trigger is being primed, but the scheduler never actually holds onto it.

I think calling unprime in __del__ will solve the problem

stuarthodgson, 2018-12-19:
Why was the change made that changed this behaviour?

eric-wieser, 2018-12-19:
To remove reference cycles in python objects that made the memory unreclaimable

stuarthodgson, 2018-12-19:
So there should only have been three objects per signal that were persistent.

eric-wieser, 2018-12-19:
That should still be true, with the caveat that they now only persist while being used.

I think something very weird is happening here - I don't see the problem on my machine running the tests, and it looks like the trigger is kept alive by the simulator anyway. It would be great if @mciepluc could post an example that causes it.

eric-wieser, 2018-12-19:
Ok, confirmed in https://api.travis-ci.org/v3/job/469643596/log.txt

eric-wieser, 2018-12-19:
Mystery solved, will make a patch tonight

mciepluc, 2018-12-20:
Thanks @eric-wieser for very quick resolution! I can confirm that both #733 and #734 resolve the problem.

eric-wieser, 2018-12-20:
Well, I broke it so felt a quick fix was necessary!

imphil, 2019-01-08:
Should be fixed now with #733 merged. @mciepluc please let me know if the current master works for you.

mciepluc, 2019-01-09:
Thanks all, issue fixed in current master

closed 2019-01-09 by mciepluc