cocotb/cocotb #731: Verifying SystemC code with OSCI Simulator

jahagirdar, 2018-12-19:
Is it possible to use cocotb to verify systemc code using the opensource OSCI simulator?

cmarqu, 2019-01-08:
I highly doubt it, since OSCI has neither VPI, VHPI, nor FLI which cocotb uses internally.

jahagirdar, 2019-01-09:
How easy/difficult is it to bolt on the subset of VPI/VHPI interface required by cocoTB to OSCI?
Can we for example, modify the  interface from iverilog or ghdl and instantiate it under sc_main()?

wallento, 2019-11-18:
Hi, I am planning to look into it. It sounds reasonable to do it. Cheers

ktbarrett, 2025-10-16:
We'll add support for OSCI when it supports us. Open a new issue if there have been developments and this is now possible.

closed 2025-10-16 by ktbarrett

cmarqu, 2026-03-11:
SCVPI needs SystemC installed which is available as a system package on Debian/Ubuntu: 
https://github.com/fvutils/scvpi/blob/34fa5e46841d31aaeed731edd39f7aedf09cbf3b/.github/workflows/ci.yaml#L36
For other systems, SystemC would have to be compiled separately.

The rest runs like this (taken from the GitHub Action run of the merged PR in the scvpi repo):

```

make: Entering directory '/home/runner/work/scvpi/scvpi/examples/basic'
=== SystemC Configuration ===
SYSTEMC: /usr
SYSTEMC_LIB_LINUX64: /usr/lib-linux64
SYSTEMC_LIB64: /usr/lib64

=== cocotb Configuration ===
COCOTB_PREFIX: 
COCOTB_LIBPYTHON: /usr/lib/x86_64-linux-gnu/libpython3.12.so.1.0
COCOTB_LIBPYTHON_DIR: /usr/lib/x86_64-linux-gnu/
COCOTB_PYTHON_BIN: /home/runner/work/scvpi/scvpi/.venv/bin/python3

=== VPI Libraries ===
COCOTB_VPI_VERILATOR: /home/runner/work/scvpi/scvpi/.venv/lib/python3.12/site-packages/cocotb/libs/libcocotbvpi_verilator.so
COCOTB_VPI_QUESTA: /home/runner/work/scvpi/scvpi/.venv/lib/python3.12/site-packages/cocotb/libs/libcocotbvpi_modelsim.so
COCOTB_VPI_XCELIUM: /home/runner/work/scvpi/scvpi/.venv/lib/python3.12/site-packages/cocotb/libs/libcocotbvpi_ius.so
COCOTB_VPI_ICARUS: /home/runner/work/scvpi/scvpi/.venv/lib/python3.12/site-packages/cocotb/libs/libcocotbvpi_icarus.vpl
make: Leaving directory '/home/runner/work/scvpi/scvpi/examples/basic'
make: Entering directory '/home/runner/work/scvpi/scvpi/examples/basic'
g++ -I/usr/include -I. -I../../src -export-dynamic ../../src/scvpi.cpp test.cpp -o test -L/usr/lib-linux64 -L/usr/lib64 -ldl -lsystemc
Running cocotb test with Icarus...

        SystemC 2.3.4-Accellera --- Apr 22 2024 14:54:19
        Copyright (c) 1996-2022 by all Contributors,
        ALL RIGHTS RESERVED
Load VPI: /home/runner/work/scvpi/scvpi/.venv/lib/python3.12/site-packages/cocotb/libs/libcocotbvpi_icarus.vpl
     -.--ns INFO     gpi                                ..mbed/gpi_embed.cpp:93   in _embed_init_python              Using Python 3.12.4 interpreter at /home/runner/work/scvpi/scvpi/.venv/bin/python3
     -.--ns INFO     gpi                                ../gpi/GpiCommon.cpp:79   in gpi_print_registered_impl       VPI registered
     0.00ns INFO     cocotb                             Running on SCVPI (SystemC) version SystemC
     0.00ns WARNING  gpi                                vpi_iterate returned NULL for type vpiInstance for object NULL
     0.00ns INFO     cocotb                             Seeding Python random module with 1771371629
     0.00ns INFO     cocotb                             Initialized cocotb v2.0.1 from /home/runner/work/scvpi/scvpi/.venv/lib/python3.12/site-packages/cocotb
Hello from TB
     0.00ns INFO     cocotb                             Running tests
     0.00ns INFO     cocotb.regression                  running tb.my_test (1/1)
my_test: HierarchyObject(abc)
  h: 
--> wait 1
my_task
<-- wait 1
--> wait 2
<-- wait 2
--> wait 3
<-- wait 3
--> wait 4
<-- wait 4
   122.00ns INFO     cocotb.regression                  tb.my_test passed
   122.00ns INFO     cocotb.regression                  **************************************************************************************
                                                        ** TEST                          STATUS  SIM TIME (ns)  REAL TIME (s)  RATIO (ns/s) **
                                                        **************************************************************************************
                                                        ** tb.my_test                     PASS         122.00           0.00     196961.16  **
                                                        **************************************************************************************
                                                        ** TESTS=1 PASS=1 FAIL=0 SKIP=0                122.00           0.00      68354.94  **
                                                        **************************************************************************************
                                                        
top::end_of_elaboration sc_argc=3

Info: /OSCI/SystemC: Simulation stopped by user.
make: Leaving directory '/home/runner/work/scvpi/scvpi/examples/basic'
make: Entering directory '/home/runner/work/scvpi/scvpi/examples/adder'
SYSTEMC=/usr
COCOTB_PYTHON_BIN=/home/runner/work/scvpi/scvpi/.venv/bin/python3
[... 110 lines cut ...]
                                                            cls.traverse(child)
                                                          File "/home/runner/work/scvpi/scvpi/.venv/lib/python3.12/site-packages/pyuvm/s09_phasing.py", line 63, in traverse
                                                            cls.execute(comp)  # first we execute this node then its children
                                                            ^^^^^^^^^^^^^^^^^
                                                          File "/home/runner/work/scvpi/scvpi/.venv/lib/python3.12/site-packages/pyuvm/s09_phasing.py", line 44, in execute
                                                            method()
                                                          File "/home/runner/work/scvpi/scvpi/examples/TinyALU_SystemC/testbench.py", line 228, in check_phase
                                                            assert passed
                                                        AssertionError: assert False
[... cut ...]
                                                        **************************************************************************************
                                                        ** TESTS=4 PASS=4 FAIL=0 SKIP=0             180003.00           0.08    2246030.12  **
                                                        **************************************************************************************
                                                        

Info: /OSCI/SystemC: Simulation stopped by user.
make: Leaving directory '/home/runner/work/scvpi/scvpi/examples/TinyALU_SystemC'
```