cocotb/cocotb #735: gpivpi: Unable to find a `gpivpi.vpi' module on the search path.

dywcheng, 2019-01-02:
Hi,

I installed with 'sudo apt-get install git make gcc g++ swig python-dev' but got the error message in the title about gpivpi when I tried any of the examples with 'make'.  I have Icarus 0.9.7 installed on my Ubuntu VM (Linux ubuntu 4.4.0-141-generic #167-Ubuntu SMP Wed Dec 5 10:40:15 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux).

Sorry for this very basic problem, but  I could not find anywhere in the doc or package on how to resolve this.

Thanks,

David

cmarqu, 2019-01-02:
Also, try running ``make`` with the ``-d`` option to get more debugging output.

dywcheng, 2019-01-03:
All three files are same as you showed.  There is no typo.  Below is a clip of the last ~100 lines of compilation output.  Do I need to install gpi from gpilab?  My Ubuntu was installed with default setting on my VMWare Workstation 12.  Thanks.

23296      Trying implicit prerequisite '/mnt/hgfs/Share2VM/TestEnv/cocotb/lib/vpi/SCCS/s.VpiImpl.cpp'.
23297      No implicit rule found for '/mnt/hgfs/Share2VM/TestEnv/cocotb/lib/vpi/VpiImpl.cpp'.
23298      Finished prerequisites of target file '/mnt/hgfs/Share2VM/TestEnv/cocotb/lib/vpi/VpiImpl.cpp'.
23299     No need to remake target '/mnt/hgfs/Share2VM/TestEnv/cocotb/lib/vpi/VpiImpl.cpp'.
23300     Considering target file '/mnt/hgfs/Share2VM/TestEnv/cocotb/lib/vpi/VpiCbHdl.cpp'.
23301      Looking for an implicit rule for '/mnt/hgfs/Share2VM/TestEnv/cocotb/lib/vpi/VpiCbHdl.cpp'.
23302      Trying pattern rule with stem 'VpiCbHdl.cpp'.
23303      Trying implicit prerequisite '/mnt/hgfs/Share2VM/TestEnv/cocotb/lib/vpi/VpiCbHdl.cpp,v'.
23304      Trying pattern rule with stem 'VpiCbHdl.cpp'.
23305      Trying implicit prerequisite '/mnt/hgfs/Share2VM/TestEnv/cocotb/lib/vpi/RCS/VpiCbHdl.cpp,v'.
23306      Trying pattern rule with stem 'VpiCbHdl.cpp'.
23307      Trying implicit prerequisite '/mnt/hgfs/Share2VM/TestEnv/cocotb/lib/vpi/RCS/VpiCbHdl.cpp'.
23308      Trying pattern rule with stem 'VpiCbHdl.cpp'.
23309      Trying implicit prerequisite '/mnt/hgfs/Share2VM/TestEnv/cocotb/lib/vpi/s.VpiCbHdl.cpp'.
23310      Trying pattern rule with stem 'VpiCbHdl.cpp'.
23311      Trying implicit prerequisite '/mnt/hgfs/Share2VM/TestEnv/cocotb/lib/vpi/SCCS/s.VpiCbHdl.cpp'.
23312      No implicit rule found for '/mnt/hgfs/Share2VM/TestEnv/cocotb/lib/vpi/VpiCbHdl.cpp'.
23313      Finished prerequisites of target file '/mnt/hgfs/Share2VM/TestEnv/cocotb/lib/vpi/VpiCbHdl.cpp'.
23314     No need to remake target '/mnt/hgfs/Share2VM/TestEnv/cocotb/lib/vpi/VpiCbHdl.cpp'.
23315     Pruning file '/mnt/hgfs/Share2VM/TestEnv/cocotb/build/libs/x86_64'.
23316    Finished prerequisites of target file '/mnt/hgfs/Share2VM/TestEnv/cocotb/build/libs/x86_64/libvpi.so'.
23317    Prerequisite '/mnt/hgfs/Share2VM/TestEnv/cocotb/lib/vpi/VpiImpl.cpp' is older than target '/mnt/hgfs/Share2VM/TestEnv/cocotb/build/libs/x86_64/libvpi.so'.
23318    Prerequisite '/mnt/hgfs/Share2VM/TestEnv/cocotb/lib/vpi/VpiCbHdl.cpp' is older than target '/mnt/hgfs/Share2VM/TestEnv/cocotb/build/libs/x86_64/libvpi.so'.
23319    Prerequisite '/mnt/hgfs/Share2VM/TestEnv/cocotb/build/libs/x86_64' is order-only for target '/mnt/hgfs/Share2VM/TestEnv/cocotb/build/libs/x86_64/libvpi.so'.
23320   No need to remake target '/mnt/hgfs/Share2VM/TestEnv/cocotb/build/libs/x86_64/libvpi.so'.
23321  Finished prerequisites of target file 'results.xml'.
23322 Must remake target 'results.xml'.
23323 PYTHONPATH=/mnt/hgfs/Share2VM/TestEnv/cocotb/build/libs/x86_64:/mnt/hgfs/Share2VM/TestEnv/cocotb:/mnt/hgfs/Share2VM/TestEnv/cocotb/examples/endian_swapper/tests:/mnt/hgfs      /Share2VM/TestEnv/cocotb/examples/endian_swapper/tests/../cosim: LD_LIBRARY_PATH=/mnt/hgfs/Share2VM/TestEnv/cocotb/build/libs/x86_64:/mnt/hgfs/Share2VM/TestEnv/cocotb/exa      mples/endian_swapper/tests/../cosim:/mnt/hgfs/Share2VM/TestEnv/cocotb/examples/endian_swapper/tests/../cosim::/usr/lib:/usr/lib:/usr/lib:/usr/lib MODULE=test_endian_swapp      er \
23324         TESTCASE= TOPLEVEL=endian_swapper_sv TOPLEVEL_LANG=verilog \
23325         /usr/bin/vvp -M /mnt/hgfs/Share2VM/TestEnv/cocotb/build/libs/x86_64 -m gpivpi sim_build/sim.vvp
23326 Putting child 0xff6e30 (results.xml) PID 13237 on the chain.
23327 Live child 0xff6e30 (results.xml) PID 13237
23328 gpivpi: Unable to find a `gpivpi.vpi' module on the search path.
23329 VCD info: dumpfile waveform.vcd opened for output.
23330 Reaping winning child 0xff6e30 PID 13237
23331 Removing child 0xff6e30 PID 13237 from chain.
23332 Successfully remade target file 'results.xml'.
23333 make[1]: Leaving directory '/mnt/hgfs/Share2VM/TestEnv/cocotb/examples/endian_swapper/tests'
23334 Reaping winning child 0xd27820 PID 13148
23335 Removing child 0xd27820 PID 13148 from chain.
23336   Successfully remade target file 'sim'.
23337  Finished prerequisites of target file 'all'.
23338 Must remake target 'all'.

cmarqu, 2019-01-03:
Please paste the output of ``ls -l /mnt/hgfs/Share2VM/TestEnv/cocotb/build/libs/x86_64/`` here (this is your ``LIB_DIR``).
I guess there is neither a ``gpivpi.vpl`` nor a ``gpivpi.vpi`` file there, but apparently you do have a file ``libvpi.so``.
Try ``ln -sf libvpi.so /mnt/hgfs/Share2VM/TestEnv/cocotb/build/libs/x86_64/gpivpi.vpl`` and run ``make`` again (no need for the ``-d`` this time). Any difference now?

cmarqu, 2019-01-03:
Sorry, I forgot to answer your other question: no, you don't need gpilab installed.

dywcheng, 2019-01-03:
It went to the next step after I moved the directory to non-shared folder between Windows and Linux and executed the ln -sf command.  (ln reported error when ran in shared folder).

Now it says it can't find simulator module.  I tried to add the lib path to PYTHONPATH but that didn't help.  I have both Python 2.7.12 and 3.5.2 installed.  Thanks.

PYTHONPATH=/home/dywcheng/nonShareVM/cocotb/build/libs/x86_64:/home/dywcheng/nonShareVM/cocotb:/home/dywcheng/nonShareVM/cocotb/examples/endian_swapper/tests:/home/dywcheng/nonShareVM/cocotb/examples/endian_swapper/tests/../cosim: LD_LIBRARY_PATH=/home/dywcheng/nonShareVM/cocotb/build/libs/x86_64:/home/dywcheng/nonShareVM/cocotb/examples/endian_swapper/tests/../cosim:/home/dywcheng/nonShareVM/cocotb/examples/endian_swapper/tests/../cosim::/usr/lib:/usr/lib:/usr/lib:/usr/lib MODULE=test_endian_swapper \
        TESTCASE= TOPLEVEL=endian_swapper_sv TOPLEVEL_LANG=verilog \
        /usr/bin/vvp -M /home/dywcheng/nonShareVM/cocotb/build/libs/x86_64 -m gpivpi sim_build/sim.vvp   
Putting child 0x1879f70 (results.xml) PID 15913 on the chain.
Live child 0x1879f70 (results.xml) PID 15913 
     -.--ns INFO     cocotb.gpi                                  gpi_embed.c:111  in embed_init_python               Did not detect Python virtual environment. Using system-wide Python interpreter.
     -.--ns INFO     cocotb.gpi                                GpiCommon.cpp:91   in gpi_print_registered_impl       VPI registered
Traceback (most recent call last):
  File "/home/dywcheng/nonShareVM/cocotb/cocotb/__init__.py", line 40, in <module>
    import cocotb.handle
  File "/home/dywcheng/nonShareVM/cocotb/cocotb/handle.py", line 45, in <module>
    import simulator
ImportError: No module named simulator
Failed to load "cocotb"
VCD info: dumpfile waveform.vcd opened for output.
Reaping winning child 0x1879f70 PID 15913 
Removing child 0x1879f70 PID 15913 from chain.
Successfully remade target file 'results.xml'.
make[1]: Leaving directory '/home/dywcheng/nonShareVM/cocotb/examples/endian_swapper/tests'
Reaping winning child 0x11f3790 PID 15824 
Removing child 0x11f3790 PID 15824 from chain.
  Successfully remade target file 'sim'.
 Finished prerequisites of target file 'all'.
Must remake target 'all'.
Successfully remade target file 'all'.

cmarqu, 2019-01-03:
The error message the ``ln`` command printed would be helpful, as it may point to the root of your problems - the Makefiles symlink things around.
In your case, it could be a missing
``ln -sf libsim.so /home/dywcheng/nonShareVM/cocotb/build/libs/x86_64/simulator.so``
(see https://api.travis-ci.org/v3/job/474461920/log.txt for how it looks when everything works (not in vmware, granted).)

closed 2019-01-04 by themperek