cocotb/cocotb #736: Readthedocs commit hook broken

imphil, 2019-01-04:
According to https://readthedocs.org/projects/cocotb/builds/ the documentation isn't getting updated with the latest commits. Possibly the GitHub service deprecation is playing a role here?

@stuarthodgson or @chiggs could you have a look what's going on there? If you want me to have a look please enable the user 'imphil' in the settings for backend access on RTD.

imphil, 2019-01-09:
Now that #740 has landed the RTD setup needs a small change. If not already done, please check in the settings that the build dependencies are loaded from `documentation/requirements.txt` (it's a setting in https://readthedocs.org/dashboard/cocotb/advanced/)

@chiggs please let me know if something needs changing on the code side.

cmarqu, 2019-01-20:
Should this be a blocker for the 1.1 release?

imphil, 2019-01-20:
no

imphil, 2019-02-06:
@stuarthodgson any update on this?

imphil, 2019-03-09:
The RTD configuration is now fixed, but the commit hook is still broken. I'll let you know once that's fixed.

themperek, 2019-03-23:
@imphil It seems to work now?

imphil, 2019-07-17:
Fixed now after repository moved.

closed 2019-07-17 by imphil