cocotb/cocotb #748: Forked coro join() sometimes doesn't return returned value

Matthewar, 2019-01-09:
The following code:
```python
@cocotb.coroutine
def wait_for(clk, cycles):
  for _ in xrange(cycles):
    yield RisingEdge(clk)
  raise ReturnValue("It worked!")

@cocotb.coroutine
def show_problem(dut):
  cocotb.fork(Clock(dut.clk, 2000, 'ps').start())

  short_wait = cocotb.fork(wait_for(dut.clk, 10))
  long_wait = cocotb.fork(wait_for(dut.clk, 30))

  yield wait_for(dut.clk, 20)

  a = yield short_wait.join()
  b = yield long_wait.join()
  print "Results: "+repr(a)+" - "+repr(b)
```
Outputs: `Results: <cocotb.triggers.NullTrigger object at 0x7fffeacd3050> - 'It worked!'`

Which is inconsistent and confusing behavior, the ReturnValue is sometimes provided (if join is called before the method finishes) and sometimes not (if join is called afterward). We ought to be returning the returned value in each case.

It looks like there is a somewhat easy workaround for this: replace `x = yield foo.join()` with:
```python
yield foo.join()
x = foo.retval
```

eric-wieser, 2019-01-09:
I think this might already be fixed in master as part of #723

Matthewar, 2019-01-09:
Sure, probably worth making a test case for it.

eric-wieser, 2019-01-09:
Actually, looks like I [already wrote that test](https://github.com/potentialventures/cocotb/pull/723/files#r241956719)

Matthewar, 2019-01-10:
You're right, there's a test and the functionality already works, however I added an additional specific test case too.

commit 71bb8914 (2019-01-10): Test case for consistent behaviour of join.

Ensure that yielding a forked coroutine with the join() method called on
it outputs the final returned value.
Closes #748.
closed 2019-01-10 by imphil