cocotb/cocotb #754: Additional join() tests

Matthewar, 2019-01-10:
A couple of errors occurring before #723 could probably do with test cases, but I'm not sure how to go about testing one of these:

Hanging on `yield Combine(Join(cocotb.fork(job0), Join(cocotb.fork(job1))`
- Due to how Join trigger implements `prime()`
- I don't know how to test whether something hangs or not, is there a test timeout?

A workaround for this issue was:
```python
active = []
active.append(cocotb.fork(job0))
active.append(cocotb.fork(job1))
# etc

while active:
    coro = active.pop(0)
    yield Join(coro)
```
Which had the potential, if job1 finishes before job0, to throw the error:

> 29704.00ns ERROR Send raised exception: 'RunningCoroutine' object has no attribute '_join'
File "\<path>/cocotb/decorators.py", line 118, in send
return self._coro.send(value)
File "some_Test.py", line 89, in run
yield Join(coro)
File "\<path>/cocotb/triggers.py", line 543, in Join
return coro._join

Avoided by:
```python
while active:
    coro = active.pop(0)
    if coro:
        yield Join(coro)
```
Or reportedly replacing `Join(coro)` with `coro.join()`

@eric-wieser, you wrote the fix for this, any thoughts on how to go about testing/is it worth testing these/do similar tests already exist?

eric-wieser, 2019-01-10:
Does the combine example now pass?

Matthewar, 2019-01-10:
I wrote a basic test that didn't work with:
`Failing test at simulator request before test run completion: Simulator shutdown prematurely`
But I wasn't sure if I had written it incorrectly:
```python
@cocotb.test()
def test_combine_join(dut):
    @cocotb.coroutine
    def wait_time(time):
        yield Timer(time)
    yield Combine(Join(cocotb.fork(wait_time(1))), Join(cocotb.fork(wait_time(2))))
    assert True
```

eric-wieser, 2019-01-10:
This looks like gh-637 to me. If you add a clock to the test, does it behave?

Matthewar, 2019-01-11:
I added a clock generator and now it hangs - so I guess this is still an issue.

Matthewar, 2019-01-11:
```
@cocotb.test()
def test_combine_join(dut):
    @cocotb.coroutine
    def wait_time(time):
        yield Timer(time)

    clk_gen = cocotb.fork(Clock(dut.clk, 100, 'ns').start())
    yield Combine(Join(cocotb.fork(wait_time(1))), Join(cocotb.fork(wait_time(2))))
    assert True
    clk_gen.kill()
```

eric-wieser, 2019-03-18:
I think this is duplicated by gh-852, for which I have a fix at #870

eric-wieser, 2019-03-22:
@Matthewar, can you check if this is fixed for you on master?

cmarqu, 2020-07-27:
Ping @Matthewar

Matthewar, 2020-07-27:
Sure, I'll try to run that this week.

ktbarrett, 2025-01-05:
This is fixed in recent versions.

closed 2025-01-05 by ktbarrett