cocotb/cocotb #756: How to set generic/parameter configuration

lcapossio, 2019-01-10:
For new users of CocoTB it is not clear from the documentation how to modify the DUT generics/parameters. Currently for modelsim I do it manually by setting SIM_ARGS accordingly. Is this the preferred way for every simulator ?

cmarqu, 2019-01-10:
Related: #682 / #692, #711, #281.

In addition to describing the arguments to be used for every supported simulator, documentation should also have an example of how to read the generic/parameter value in Python.

Bamban, 2019-03-29:
I upvote this issue! 
And I have a related question. GHDL has a `-gGENERIC=something` option at elaboration (`-e`) not very well documented it's true. I tried to use it with EXTRA_ARGS and COMPILE_ARGS but each variable is passed to `ghdl -a` AND `ghdl -e`. It creates an error when passed to `ghdl -e`. How can I give `ghdl -a` an exclusive option?

aairabella, 2019-07-09:
Is there any docs in this topic? I used `make sim PARAMETERS=parameter_name=parameter_value` up to v1.0, but it doesn't work in v1.1.

imphil, 2019-07-09:
cocotb doesn't handle parameters/defines/generic itself, it only allows you to pass them through to the compiler/simulator. You can use `SIM_ARGS` and `COMPILE_ARGS` for that ([as documented here](https://cocotb.readthedocs.io/en/latest/building.html#make-variables)). You need to have a look at the simulator's user guide which flags it expects, and to which tool parameters should be passed (typically to the simulation as part of SIM_ARGS, sometimes to both).

aairabella, 2019-07-09:
Well, I'm using v1.1 now and so far this it the solution I found:

`make sim EXTRA_ARGS=-p<top_level_module_name>.<parameter_name>=<parameter_value>`

I'm using Verilog and my simulator is `icarus`.

matthuszagh, 2019-10-18:
Is setting module parameter values via cocotb something that would be feasible to support? This would be useful for testing a module under different parameterizations. I guess it would be possible to define individual testcases and then conditional makefile variable assignments, but this seems more manual/verbose. Am I missing an easier way to do this?

stefan-muc, 2020-08-06:
@Bamban ```make SIM_ARGS="-gGENERIC=value"``` or add SIM_ARGS to your ```Makefile```

But yeah, everyone does it differently. ```questa``` uses an separate environment variable ```GENERICS="GENERIC=value"```` 
https://dornerworks.com/blog/setting-vhdl-generics-in-fpga-verification-made-easy-with-cocotb/

ktbarrett, 2020-08-06:
@stefan-muc Note that the Questa makefile does not support `GENERICS` anymore. You have to pass generics using Questa's syntax in `SIM_ARGS` or `COMPILE_ARGS`.

stefan-muc, 2020-08-06:
@ktbarrett Thanks for pointing out! I'm using ```ghdl```

garmin-mjames, 2020-08-06:
The Makefile in #1502 shows how to set parameters/generics for several simulators.

BobbbbbZ, 2024-06-28:
Has anyone solved the issue? Now I am also bothered by the configuration of parameter if used in icarus verilog.

ktbarrett, 2024-06-28:
What issue? If you are using the makefiles use `COMPILE_ARGS` and/or `SIM_ARGS` as detailed in this issue to pass command options for generics for your simulator. The Python runner has the `parameters` argument . Seeing as we are gently phasing out the Makefiles I see no reason to backport this feature.

closed 2024-06-28 by ktbarrett