cocotb/cocotb #760: Log summary improvements

Matthewar, 2019-01-11:
The cocotb test summary could provide additional useful information.
Current example of an output:
```
# COUT: 23183.53ns INFO     cocotb.regression                         regression.py:227  in handle_result                   Test Passed: test_3
# COUT: 23183.53ns ERROR    cocotb.regression                         regression.py:177  in tear_down                       Failed 1 out of 28 tests (0 skipped)
# COUT: 23183.53ns INFO     cocotb.regression                         regression.py:348  in _log_test_summary               ******************************************************************************************************************
# COUT:                                                                                                                     ** TEST                                                      PASS/FAIL  SIM TIME(NS)  REAL TIME(S)  RATIO(NS/S) **
# COUT:                                                                                                                     ******************************************************************************************************************
# COUT:                                                                                                                     ** test_suite.test_1                                           PASS         1150.00          1.61       712.31  **
# COUT:                                                                                                                     ** test_suite.test_2                                           PASS          373.50          0.44       840.14  **
# COUT:                                                                                                                     ** test_suite.test_3                                           PASS          376.00          0.46       819.86  **
# COUT:                                                                                                                     ******************************************************************************************************************
# COUT:                                                                                                                     
# COUT: 23183.53ns INFO     cocotb.regression                         regression.py:365  in _log_sim_summary                *************************************************************************************
# COUT:                                                                                                                     **                                 ERRORS : 1                                      **
# COUT:                                                                                                                     *************************************************************************************
# COUT:                                                                                                                     **                               SIM TIME : 23183.53 NS                            **
# COUT:                                                                                                                     **                              REAL TIME : 28.30 S                                **
# COUT:                                                                                                                     **                        SIM / REAL TIME : 819.15 NS/S                            **
# COUT:                                                                                                                     *************************************************************************************
```
This could be improved with more useful information, especially as there's lots of whitespace to the left that's wasted.
Potential improvements:
- Test start/end times (in simulator time)
    - Makes it easier to locate output in other logs/waves relating to a specific test case
    - Running individual test could be used for better correlation to waves, but sometimes the error is caused by a lingering state so running the suite is required
- Potentially tricky to implement, but showing the source line of the testbench where the failure occurred, and whether it was a `TestFailure` or some other exception

Mock-up of what this might look like:
```
# COUT: 23183.53ns INFO     cocotb.regression                         regression.py:348  in _log_test_summary               ******************************************************************************************************************
# COUT:                                                                                                                     ** TEST                                                      PASS/FAIL  SIM TIME(NS)  REAL TIME(S)  RATIO(NS/S) **
# COUT: Run Times:          Notes:                                                                                          ******************************************************************************************************************
# COUT: 12345-34567ns                                                                                                       ** test_suite.test_1                                           PASS          251.00          0.34       733.61  **
# COUT: 34568-43210ns       TestFailure @ test_suite.py:123                                                                 ** test_suite.test_2                                           FAIL         1015.50          1.50       678.36  **
# COUT: 43211-55555ns       ValueError  @ test_suite.py:564                                                                 ** test_suite.test_3                                           ERROR         449.50          0.56       804.19  **
# COUT:                                                                                                                     ******************************************************************************************************************
# COUT:                                                                                                                     
# COUT: 23183.53ns INFO     cocotb.regression                         regression.py:365  in _log_sim_summary                *************************************************************************************
# COUT:                                                                                                                     **                                 ERRORS : 1                                      **
# COUT:                                                                                                                     *************************************************************************************
# COUT:                                                                                                                     **                               SIM TIME : 23183.53 NS                            **
# COUT:                                                                                                                     **                              REAL TIME : 28.30 S                                **
# COUT:                                                                                                                     **                        SIM / REAL TIME : 819.15 NS/S                            **
# COUT:                                                                                                                     *************************************************************************************
# COUT:                                                                                                                     
```
Probably worth a discussion of what others think would be useful features here.

cmarqu, 2019-01-11:
See also #10.

eric-wieser, 2019-10-30:
> but showing the source line of the testbench where the failure occurred

I believe this is done, but possibly only for exceptions and not failures.

eric-wieser, 2020-05-08:
Nevermind, I see this is specifically about the summary table.