cocotb/cocotb #765: Handle X values in BinaryValue()

Matthewar, 2019-01-13:
Continuing from #750 - need to add handling of X values in the following functions in `BinaryValue()` class:
- `__iand__`
- `__ixor__`
- `__ior__`
- `__hex__`
- `__oct__`

eric-wieser, 2019-01-13:
In my opinion, the `__iop__` overloads are a mistake, and we should not have added them in the first place.

The problem they have is something like:
```
a = BinaryValue(1)
b = a
a += 1
assert b.integer == 2 
assert b.integer == 1   # fails if we implement __iadd__, passes if we don't
```

Regarding `__hex__` and `__oct__` - these methods are removed in python 3, and python 2 is EOL at the end of the year - so it may not be worth spending the time to implement them.

ktbarrett, 2024-05-18:
BinaryValue was replaced with LogicArray and it should not have issues with Xs or Zs in bitwise logic operators.

closed 2024-05-18 by ktbarrett