cocotb/cocotb #767: Dropping Python 2 support

eric-wieser, 2019-01-13:
Upstream support for Python 2.7 [ends on 2020-01-01](https://pythonclock.org/).

I know it's a long way off right now, and the cocotb release cycle is in its infancy - but at some point it would be good for cocotb to decide if and/or when to make its final Python 2.7 release, as [many other projects are already doing](https://python3statement.org/).

Creating this issue now to act as a catalyst for this discussion - it doesn't need to be acted on any time soon.

Matthewar, 2019-01-26:
Potentially somewhat related to #610.
Also, while Python 2.7 is reaching end of its upstream support, are enterprise solutions (such as RHEL7) [promising support until the end of their lives](https://access.redhat.com/documentation/en-us/red_hat_enterprise_linux/7/html/7.5_release_notes/chap-red_hat_enterprise_linux-7.5_release_notes-deprecated_functionality)?
That doesn't mean that we can't start utilising Python3 features such as for the scheduler (so long, as discussed in that issue, old testbench syntax still works) but perhaps a branch would remain with Python2 backported fixes?

stuarthodgson, 2019-01-26:
There seem to be a lot of outstanding issues and features people want. Dropping python 2 seems quite low down. It's not as if 3 does not work, so why the push?   Even when 2 maintenance ends it's not like world will stop using it.

eric-wieser, 2019-12-03:
I'd propose we make 1.3 the last minor release to support Python 2.

It's clear to me that 1.2 (already released) cannot be the last release for Python 2. There are many fixes for Python 2 in master, and it would be foolish to never release them - even if the release does come after the Python 2.7 EOL date. I don't think there's much need to discuss this here.

The more controversial claim I'm making here, and the one I'd like responses to, is that we should make cocotb 1.4.x support only Python >= 3.5.

ktbarrett, 2019-12-03:
Perhaps we can please all parties by instead of going to cocotb 1.4.x and Python 3 after this release, move to cocotb 2.0; allowing a cocotb 1.x branch to stay for long-term Python 2 support. Those that are stuck or unwilling to upgrade to Python 3 will still be allowed a place to contribute back bugfixes for Python 2, or even backport features from the 2.x branch if they are particularly stubborn. I don't think we need to put as much resources into maintaining the 1.x branch as developing the 2.x branch, but there should be at least one maintainer who is willing to look at 1.x issues.

The [Catch](https://github.com/catchorg/Catch2) project does this.

hugovk, 2019-12-03:
> The more controversial claim I'm making here, and the one I'd like responses to, is that we should make cocotb 1.4.x support only Python >= 3.5.

Not controversial at all, Python 3.4 has been EOL since March, and is no longer receiving security updates (or any updates) from the core Python team.

Version | Release date | Supported until
--- | ---------- | ----------
2.7 | 2010-07-03 | 2020-01-01
3.4 | 2014-03-16 | 2019-03-16

Source: https://en.wikipedia.org/wiki/CPython#Version_history

It's also little used. Here's the pip installs for cocotb from PyPI for November 2019:

| category | percent | downloads |
|----------|--------:|----------:|
| 3.5      |  45.14% |       655 |
| 3.7      |  33.63% |       488 |
| 2.7      |   7.99% |       116 |
| 3.6      |   6.13% |        89 |
| null     |   4.07% |        59 |
| 3.8      |   2.83% |        41 |
| 3.4      |   0.14% |         2 |
| 2.6      |   0.07% |         1 |
| Total    |         |     1,451 |

Date range: 2019-11-01 - 2019-11-30

Source: `pip install -U pypistats && pypistats python_minor cocotb --last-month`

eric-wieser, 2019-12-03:
@hugovk: To be clear, cocotb already does not support python 3.4. The relevant part of that claim is that 2.7 is less than 3.5 too.

eric-wieser, 2019-12-03:
@ktbarrett:

> but there should be at least one maintainer who is willing to look at 1.x issues.

Are you saying "we need to ensure this is true", or "we should not consider having a 1.x at all unless this is true"?

hugovk, 2019-12-03:
(Edited previous post, sorry for confusion!)

Back on topic: 8% for Python 2.7 is very low, considering many projects with a higher share are also dropping 2.7 for or before the EOL on 2020-01-01.

* https://dev.to/hugovk/python-version-share-over-time-5-7ik
* https://python3statement.org/

It sounds sensible to make a final release supporting 2.7 soon with the current code, and then drop 2.7 for the following release.

About maintenance branch, here's pytest's plan (in short: they've already dropped 2.7 from mainstream; will backport bugs until EOL; and will accept community fixes and release until mid-2020).

* https://docs.pytest.org/en/latest/py27-py34-deprecation.html

ktbarrett, 2019-12-03:
@eric-wieser 
>"we should not consider having a 1.x at all unless this *could be* true"

This. I'd rather not encourage people to make PRs against the 1.x branch if they are just going to sit there because there is no maintainer willing to do reviews and bring in fixes.

@hugovk 
The fact that 2.7 is over 5% and has more adoption than a supported version of 3 feels like a lot to me, and tells me that we shouldn't just burn that bridge. And it's likely that number will not decrease, even after EOL, if they are still using it at this point. I think pytest is a bad example, it's a very popular, very mature project. Cocotb is not. The current version of pytest will likely work very well forever. Cocotb I'm not so confident about.

Note, I'm not advocating active development of Python 2. Let it die a **peaceful** death.

hugovk, 2019-12-03:
Dropping 2.7 in a newer release does not stop people with 2.7 using an older release. They are still available on PyPI.

And the `python_requires` parameter in `setup.py` helps pip install the newest available version for the user's running Python.

ktbarrett, 2019-12-04:
Yes, of course older releases will still be available. I just want to support community-driven bugfixes and backports for those that are going to continue to use Python 2 past EOL. And all that requires is a branch.

eric-wieser, 2020-01-18:
This was resolved, xref https://github.com/cocotb/cocotb/pull/1291#issuecomment-574623212

closed 2020-01-18 by eric-wieser

s3bs, 2020-02-22:
This seems a decision that might work for small projects or university use. 
Years of simulation code and co simulation code in a corporate environment with lots of py27 dependencies makes it way too costly and time consuming to move forward to py3+

themperek, 2020-02-22:
@s3bs Nothing is lost. You are welcome to use version 1.3 and contribute bug fixes.

eric-wieser, 2020-02-22:
@s3bs: As maintainers, improving stability and adding new features for while supporting python27 is both time consuming and error-prone, and with limited maintainer time we can't afford to pay that forever. Choosing to stay on Python 2.7 leaves us unable to build upon the last 10(!) years of work by the python core developers.

As @themperek points out, there is nothing stopping you from using cocotb 1.3 on python 2.7 forever - but just like you are very unlikely to get any more bugfixes for Python 2.7 itself, you are also unlikely to get any new bugfixes for cocotb.