cocotb/cocotb #768: setimmediatevalue doesnt work for signals wider than 32 bit

jahagirdar, 2019-01-14:
[cocotb_test.tar.gz](https://github.com/potentialventures/cocotb/files/2756337/cocotb_test.tar.gz)

See attached testcase for details.

It looks like setimmediatevalue does not take effect if the signal width is greater than 32.

I tried the following variation and all of them fail

```
        dut.inp_short63.setimmediatevalue(0)
        dut.inp_short64.setimmediatevalue(BinaryValue(value=0,bits=65).value)
        dut.inp.setimmediatevalue(BinaryValue(value='0x0', bits=1024))
```

The normal assignment statement works
```
dut.inp_short70 <= BinaryValue(value=0xffffffffffffffff,bits=71).value
```

jahagirdar, 2019-01-16:
I was facing this problem with an older version( pulled around a year back) and did a pull and verified that it occurs with the current version too before raising the ticket.

This is the first time that I am using setimmediatevalue with multibit control signals.

So I would classify it as a bug.

imphil, 2019-01-16:
Thanks for confirming that!

eric-wieser, 2019-01-28:
Tested by modifying the `issue_142` test, doesn't reproduce for me:
```python
@cocotb.test()
def issue_142_overflow_error(dut):
    """Tranparently convert ints too long to pass
       through the GPI interface natively into BinaryValues"""
    cocotb.fork(Clock(dut.clk, 2500).start())

    def _compare(value):
        if int(dut.stream_in_data_wide.value) != int(value):
            raise TestFailure("Expecting 0x%x but got 0x%x on %s" % (
                int(value), int(dut.stream_in_data_wide.value),
                str(dut.stream_in_data_wide)))

    values = [
        0,
        0x7FFFFFFF,
        0x7FFFFFFFFFFF,
        BinaryValue(0x7FFFFFFFFFFFFF,len(dut.stream_in_data_wide),bigEndian=False),
        BinaryValue(0, bits=len(dut.stream_in_data_wide))
    ]

    # Wider values are transparently converted to BinaryValues
    for value in values:

        dut.stream_in_data_wide <= value
        yield RisingEdge(dut.clk)
        _compare(value)

    for value in values:
        dut.stream_in_data_wide = value
        yield RisingEdge(dut.clk)
        _compare(value)

    for value in values:
        dut.stream_in_data_wide.setimmediatevalue(value)
        yield RisingEdge(dut.clk)
        _compare(value)
``

eric-wieser, 2019-01-28:
Ah, this is probably #637 in a new and exciting form.

It only fails if the write happens before any events have been yielded.

This fails, but passes if `do_wait` is changed to `True`
```python
import cocotb
from cocotb.triggers import Timer, RisingEdge, ReadOnly
from cocotb.regression import TestFactory

@cocotb.coroutinue()
def do_test(dut, do_wait=False):
    if do_wait:
        yield Timer(1)
    dut.inp.setimmediatevalue(0)
    yield RisingEdge(dut.done)
    assert dut.inp.value == 0
    yield ReadOnly()
```

eric-wieser, 2019-01-28:
Here's the bug:
```
// Value related functions
int VpiSignalObjHdl::set_signal_value(long value)
{
    FENTER
    s_vpi_value value_s;

    value_s.value.integer = value;
    value_s.format = vpiIntVal;

    s_vpi_time vpi_time_s;

    vpi_time_s.type = vpiSimTime;
    vpi_time_s.high = 0;
    vpi_time_s.low  = 0;

    // Use Inertial delay to schedule an event, thus behaving like a verilog testbench
    vpi_put_value(GpiObjHdl::get_handle<vpiHandle>(), &value_s, &vpi_time_s, vpiInertialDelay);
    check_vpi_error();

    FEXIT
    return 0;
}

int VpiSignalObjHdl::set_signal_value(double value)
{
    FENTER
    s_vpi_value value_s;

    value_s.value.real = value;
    value_s.format = vpiRealVal;

    s_vpi_time vpi_time_s;

    vpi_time_s.type = vpiSimTime;
    vpi_time_s.high = 0;
    vpi_time_s.low  = 0;

    vpi_put_value(GpiObjHdl::get_handle<vpiHandle>(), &value_s, &vpi_time_s, vpiInertialDelay);
    check_vpi_error();

    FEXIT
    return 0;
}

int VpiSignalObjHdl::set_signal_value(std::string &value)
{
    FENTER
    s_vpi_value value_s;

    std::vector<char> writable(value.begin(), value.end());
    writable.push_back('\0');

    value_s.value.str = &writable[0];
    value_s.format = vpiBinStrVal;

    vpi_put_value(GpiObjHdl::get_handle<vpiHandle>(), &value_s, NULL, vpiNoDelay);
    check_vpi_error();

    FEXIT
    return 0;
}
```

closed 2019-02-05 by imphil