cocotb/cocotb #774: Document C code

cmarqu, 2019-01-20:
We should make documentation of the C code in ``cocotb/lib/`` available in Sphinx.
There are incomplete Doxygen annotations (see #8) in some of the files, so we might want to use https://breathe.readthedocs.io/en/latest/ (a Doxygen -> Sphinx bridge).

Some other links/examples:
* http://www.sphinx-doc.org/en/master/usage/restructuredtext/domains.html#cross-referencing-c-constructs
* https://docs.scipy.org/doc/numpy-1.15.0/reference/c-api.types-and-structures.html
   * https://raw.githubusercontent.com/sunny11mind/NumPy/master/doc/source/reference/c-api.types-and-structures.rst

wallento, 2019-01-21:
I like the idea. Just to add to the examples: https://github.com/optimsoc/optimsoc/tree/master/doc/api

IIRC, breathe is broken and/or unmaintained with recent sphinx, but its easy to workaround..

wallento, 2019-01-22:
Yeah, I can have a look at building the basic infrastructure the next days. I am not too deep into the C code,  but once the stubs are there, people can hopefully volunteer filling them :)

cmarqu, 2019-02-12:
I have started looking at this, and while ``optimsoc-apidoc`` didn't produce any output (it tries to extract only "groups" which cocotb apparently doesn't have?), breathe itself seemed to work fine.
Playing with some options is still needed, will do that soon.

wallento, 2019-02-13:
yeah, its pretty much narrowed to our use case. Thanks for looking into it. I cannot fully remember what the issue with breathe was, but I roughly remember it was some missing updates to it to support a newer Sphinx version, which we needed for another feature. So this may be solved.

Sorry, I am snowed currently..

cmarqu, 2019-02-18:
Okay, this lives in https://github.com/cmarqu/cocotb/tree/doc-c-library2 for now.
One thing that might be considered a drawback is that all these C++ classes and their methods produce quite a lot of entries in Sphinx' index page, drowning out the "end user" Python entries - but even Sphinx' own index looks like this.

cmarqu, 2019-03-03:
I now have an example rendering here: https://cocotb-cmarqu.readthedocs.io/en/doc-c-library2/library_reference_c.html (the index: https://cocotb-cmarqu.readthedocs.io/en/doc-c-library2/genindex.html)

cmarqu, 2019-06-27:
This is now live at https://cocotb.readthedocs.io/en/latest/library_reference_c.html

closed 2019-06-27 by cmarqu