cocotb/cocotb #779: binary.py: Issue truncation warning only when required

jahagirdar, 2019-01-24:
Currently whenever BinaryValue is called with num_bits smaller than the length of the input binstr
We get a warning of 
   
`   WARNING truncating value to match requested number of bits  (64)
`

This warning is seen even if the integer value post truncation is same as the original value. The warning would be useful if it was issued only when truncation resulted in a change of value.

e.g. 

```
import cocotb
from cocotb.decorators import coroutine
from cocotb.triggers import Timer, RisingEdge, ReadOnly
from cocotb.binary import BinaryValue
from cocotb.regression import TestFactory

@cocotb.coroutine
def run_dma_test(dut):

    b=BinaryValue(value=8, bits=32, bigEndian=False)
    c=BinaryValue(value=b.binstr, bits=8, bigEndian=False)
    cocotb.log.info("Value is" + repr(c.hex()))
    yield RisingEdge(dut.done)
    yield ReadOnly()

factory = TestFactory(run_dma_test)
factory.generate_tests()
```

Results in 

WARNING: truncating value to match requested number of bits (32 -> 8)
    0.00ns INFO     Value is'0x8'

The warning would have been more useful if it was emitted only when truncation changed value 
e.g. 
if b=0xa0b3 and trucation resulted in c=0xb3

jahagirdar, 2019-01-26:
I can see 3 possible way's to implement the proposed truncate operation.

1. Pass ```truncate =False``` as an optional parameter to __init__; If False the current behavior is seen, If True Warning is issues only if truncation leads to loss of value.
2. Add a new function ```truncate``` to binary.py which replicates the current adjust routine without the warning, or which acts as a wrapper over them and supresses the warning.
3. Leave binary.py as is and implement truncation in my code. 

Which of the 3 option would be most suitable.

ktbarrett, 2024-05-18:
BinaryValue is dead. The new behavior is not to truncate hut to error.

closed 2024-05-18 by ktbarrett