cocotb/cocotb #781: Icarus Makefile uses options not available in icarus 0.9.7

imphil, 2019-01-25:
7b46fd5 broke builds on ubuntu 16.04, where the distribution iverilog does not support -g2012.

Some observations:
* Icarus 0.9.7 has been released on 2013-08-21.
* The "-g2012" flag was added in Oct 2016 and is first release in iverilog 10.2 (August 2017). 
* Our tests require iverilog 10.2
* CI only tests 10.2

Our options:
1. Require iverilog 10.2 (document that)
2. Do version detection in the Makefile
3. Don't always pass -g2012 in the Makefile, but let users choose.

We had option 3 before, that's giving a pretty bad user experience, since SV2012 is used really a lot these days. Option 2 could be done. I personally would just go with option 1.

Matthewar, 2019-01-25:
While SV2012 is supported in Icarus, I don't think it's fully supported, so it may not matter if we change it to an older flag, e.g. g2005-sv? I think it was added during the review stage of the PR to get the CI tests passing, since a new test used queues - however Icarus doesn't support queues properly so I changed the module to use a shift register fifo instead https://github.com/potentialventures/cocotb/blob/master/tests/designs/avalon_streaming_module/avalon_streaming.sv

imphil, 2019-01-25:
I wouldn't expect g2005-sv to be supported in 0.9.7 either, but please double-check.

Matthewar, 2019-01-25:
Looks like `-g2005-sv` was added in tag `v10_2` so I guess not. Do our tests and makefiles need to continue to support a 5/6 year old version? If so, we could remove the flag and change the system verilog test to verilog.
If you think people should be able to continue to use the makefiles for this icarus version, but the tests don't have to work for them, we could add documentation stating they require a certain version (or an assert in the test makefile) and change the `COMPILE_ARGS` to optionally add `g2012` if you haven't set it yourself in the simulator makefile.
Or as you suggest in option 1, we could just state that iverilog must be version 10.2.

I think because we don't test in CI any older versions we can't guarantee that 0.9.7 is still supported, and should definitely document this somewhere no matter what we decide. But I don't really mind about changing the tests/makefile too.

eric-wieser, 2019-01-28:
> We had option 3 before, that's giving a pretty bad user experience, since SV2012 is used really a lot these days. 

What happens if a user wants to explicitly _not_ use sv2012? Can they override the flag?

cmarqu, 2019-01-29:
According to https://stackoverflow.com/a/32696474/41347 I'd expect that ``make COMPILE_ARGS="..."`` clears even the ``+=`` assignment, but the user would have to specify all the other needed ``COMPILE_ARGS`` which is a nuisance (and would have to figure out what to set them to in the first place).
Something like
```
SV_VERSION ?= -g2012
COMPILE_ARGS += $(SV_VERSION)
```
is better I guess.
(Maybe we really should move away from Makefiles to something that has proper command line handling and config files. That way, we could better support rare cases like per-file version options etc.)

ktbarrett, 2025-03-27:
We no longer support either Ubuntu 16.04 or Icarus < 10.3.

closed 2025-03-27 by ktbarrett