cocotb/cocotb #804: Replace license header with SPDX identifier

imphil, 2019-02-06:
Currently we have a large comment block at every source file containing the 3-clause BSD license header. I'm proposing to strip this down to a reference to the LICENSE file and a SPDX license identifier (28 -> 3 lines).

Currently we have:
```
###############################################################################
# Copyright (c) 2013 Potential Ventures Ltd
# Copyright (c) 2013 SolarFlare Communications Inc
# All rights reserved.
#
# Redistribution and use in source and binary forms, with or without
# modification, are permitted provided that the following conditions are met:
#     * Redistributions of source code must retain the above copyright
#       notice, this list of conditions and the following disclaimer.
#     * Redistributions in binary form must reproduce the above copyright
#       notice, this list of conditions and the following disclaimer in the
#       documentation and/or other materials provided with the distribution.
#     * Neither the name of Potential Ventures Ltd,
#       SolarFlare Communications Inc nor the
#       names of its contributors may be used to endorse or promote products
#       derived from this software without specific prior written permission.
#
# THIS SOFTWARE IS PROVIDED BY THE COPYRIGHT HOLDERS AND CONTRIBUTORS "AS IS" AND
# ANY EXPRESS OR IMPLIED WARRANTIES, INCLUDING, BUT NOT LIMITED TO, THE IMPLIED
# WARRANTIES OF MERCHANTABILITY AND FITNESS FOR A PARTICULAR PURPOSE ARE
# DISCLAIMED. IN NO EVENT SHALL POTENTIAL VENTURES LTD BE LIABLE FOR ANY
# DIRECT, INDIRECT, INCIDENTAL, SPECIAL, EXEMPLARY, OR CONSEQUENTIAL DAMAGES
# (INCLUDING, BUT NOT LIMITED TO, PROCUREMENT OF SUBSTITUTE GOODS OR SERVICES;
# LOSS OF USE, DATA, OR PROFITS; OR BUSINESS INTERRUPTION) HOWEVER CAUSED AND
# ON ANY THEORY OF LIABILITY, WHETHER IN CONTRACT, STRICT LIABILITY, OR TORT
# (INCLUDING NEGLIGENCE OR OTHERWISE) ARISING IN ANY WAY OUT OF THE USE OF THIS
# SOFTWARE, EVEN IF ADVISED OF THE POSSIBILITY OF SUCH DAMAGE.
###############################################################################
```

I'm proposing to change this to:
```
# Copyright cocotb contributors
# Licensed under the Revised BSD License, see LICENSE for details.
# SPDX-License-Identifier: BSD-3-Clause
```

Motivations:
- Reduce the boilerplate at the beginning of files
- Optional: Remove the year to avoid updating and merge conflicts.

Things to discuss, especially for @stuarthodgson and @chiggs:

- According to many lawyers it's safe to remove the year from the copyright statements (including large corporate ones). Let me know if you have differing requirements here.

- Is `Copyright cocotb contributors` sufficient for you @stuarthodgson and @chiggs, or should this be `Copyright Potential Ventures Ltd, SolarFlare Communications Inc, and cocotb contributors`, or something along those lines?

stuarthodgson, 2019-02-06:
Will have to have a think about this some more. Given the rate of file addition and that we don't modify it after creation I'm not sure is much of an issue at the moment anyway.

cmarqu, 2019-02-10:
I had added 2018/2019 as part of my PRs because it seems to me that general consensus in many projects is to update the year (if present in the first place) whenever "substantial" changes are made to a file.

#798 now doesn't add "2019" anymore; I can remove the "2018" added in previously merged PRs if wanted.

ktbarrett, 2025-03-27:
Closed by #4576.

closed 2025-03-27 by ktbarrett