cocotb/cocotb #811: Remove FIXMEs from docstrings

cmarqu, 2019-02-10:
In https://github.com/potentialventures/cocotb/pull/798/commits/1cfeeb02472fb0d33a1a04ef521703e55f67a429, I introduced some FIXMEs in docstring for methods in ``triggers.py`` (https://github.com/potentialventures/cocotb/blob/1cfeeb02472fb0d33a1a04ef521703e55f67a429/cocotb/triggers.py).
Someone knowledgeable should properly document these methods.

eric-wieser, 2019-07-01:
Easy documentation for `prime`: "Don't call this in your own code, this is for internal by the scheduler"

cmarqu, 2019-07-01:
Okay :) Internal enough so that we should rename this to ``_prime()``?

eric-wieser, 2019-07-01:
Possibly. `prime` is public API if you're writing your own scheduler :)

cmarqu, 2020-07-27:
This is fixed now.

closed 2020-07-27 by cmarqu