cocotb/cocotb #812: Log instantiation naming

jeinstei, 2019-02-11:
Low priority, but all the examples call dut._log instead of dut.log, which would be more in line with Python naming conventions. Could we create the dut.log interface instead of _log, if it is supposed to be called for logging by the user?

cmarqu, 2019-02-12:
This is supposed to prevent naming collisions, since ``dut.<something>`` could be an existing signal or instance etc. in the design.

(That said, PEP 8 actually recommends using a _trailing_ underscore for such usage instead of a leading one.)

jeinstei, 2019-02-12:
Point taken. I suppose this is more of a philosophy question then, if there should be a cocotb data structure underneath dut that holds everything such as the log? dut.cocotb.log, for example? Obviously this would be a change, but the likelihood of someone naming a signal cocotb is low, and it can be tested for.

eric-wieser, 2020-01-15:
> which would be more in line with Python naming conventions.

Note that `namedtuple._replace` is in the standard library, with exactly the same naming convention, again motivated by avoiding collisions. I think we should just make sure to explain the leading underscore in the documentation.

ktbarrett, 2025-03-27:
These loggers are now private. But more generally, attributes on `SimHandleBase` and `HierarchyObjectBase` will start with underscores to prevent naming collisions, and they are listed as public in the documentation.

closed 2025-03-27 by ktbarrett