cocotb/cocotb #815: '<=' not supported between instances of 'HierarchyObject' and 'int'

andresdemski, 2019-02-18:
Hello,
I upgraded GHDL to
```
GHDL 0.36-dev (20181129-170-gf8437ed8) [Dunoon edition]
 Compiled with GNAT Version: 8.2.1 20181127
 llvm code generator
Written by Tristan Gingold.
```
I tried to run a test (it was working before the upgrate) And i get the following error:
```
Send raised exception:
Traceback (most recent call last):
  File "/home/ademski/Documents/satellogic/hdlteam/cocotb/cocotb/decorators.py", line 121, in send
    return self._coro.send(value)
  File "/home/ademski/Documents/satellogic/hdlteam/cocotb/cocotb/clock.py", line 64, in start
    self.signal <= 1
TypeError: '<=' not supported between instances of 'HierarchyObject' and 'int'
```
I fork the clock coroutine with the following line:
```python
cocotb.fork(Clock(dut.clk, CLK_PERIOD, units=CLK_PERIOD_UNITS).start())
```
I tried doing `make clean` and rebuilding vhpi but it doesn't work.

Thx
Andres

cmarqu, 2019-02-18:
If you try to ``dut.clk <= 1``, does that raise the same exception?

andresdemski, 2019-02-19:
yep, that raise the same exception:
```
 Traceback (most recent call last):
   File "/home/ademski/<COCOPATH>/cocotb/cocotb/decorators.py", line 220, in send
     return self._coro.send(value)
   File "/home/ademski/<TESTPATH>/test.py", line 164, in test
     dut.clk <= 1
 TypeError: '<=' not supported between instances of 'HierarchyObject' and 'int'
```

cmarqu, 2019-02-19:
I would now try narrow this down more - like does it happen with another "regular" signal too, or is clk somehow special in your design (do you also have something else in the design that is called clk?), or is GHDL in that version behaving strangely?
Maybe also ``test_discovery`` can give some clues.

andresdemski, 2019-02-19:
Here i dockerize the "test_case".
 
[error.zip](https://github.com/potentialventures/cocotb/files/2880457/error.zip)

Execution cmd:
```bash
mkdir <path> && unzip ~/Downloads/error.zip && cd ./coco_error/ && ./runsim.sh
```

andresdemski, 2019-02-19:
> I would now try narrow this down more - like does it happen with another "regular" signal too, or is clk somehow special in your design (do you also have something else in the design that is called clk?), or is GHDL in that version behaving strangely?
Clk exist, and the error must be other if it is the reason (no attribute called dut.clk)

> Maybe also `test_discovery` can give some clues.
Running inside docker:

## test_dicovery with iverilog
```bash
make -C /cocotb/tests/test_cases/test_discovery/ TOPLEVEL_LANG=verilog COCOTB_REDUCED_LOG_FMT=true
```
Result: PASS
## test_dicovery with ghdl
```bash
make -C /cocotb/tests/test_cases/test_discovery/ TOPLEVEL_LANG=vhdl SIM=ghdl COCOTB_REDUCED_LOG_FMT=true
```
Result: FAILED

```    0.00ns INFO     Running on GHDL version 0.1
    0.00ns INFO     Python interpreter initialised and cocotb loaded!
    0.00ns INFO     Running tests with Cocotb v1.1.0 from /cocotb
    0.00ns INFO     Seeding Python random module with 1550588767
    0.00ns CRITICAL Failed to import module test_discovery: No module named 'test_discovery'
    0.00ns INFO     MODULE variable was "test_discovery.test_vhdl_indexed_name"
    0.00ns INFO     Traceback: 
    0.00ns INFO     Traceback (most recent call last):
                      File "/cocotb/cocotb/regression.py", line 133, in initialise
                        module = _my_import(module_name)
                      File "/cocotb/cocotb/regression.py", line 63, in _my_import
                        mod = __import__(name)
                    ImportError: No module named 'test_discovery'
```

andresdemski, 2019-02-19:
Using `debian:buster` image and `ghdl-gcc` package, my test_case works. The problem is the interaction with the "new ghdl".

andresdemski, 2019-02-19:
Could be relater to #818 ?

cmarqu, 2019-02-19:
> Could be relater to #818 ?

This code is unchanged since 4 years: https://github.com/potentialventures/cocotb/blame/db2f4e02ae3f788218a2b915f16016b54c52b6eb/lib/vpi/VpiImpl.cpp#L137

cmarqu, 2019-02-19:
Being able to ``git bisect`` GHDL would be helpful here.

andresdemski, 2019-02-19:
The error happens with current ghdl master branch (https://github.com/ghdl/docker). I can do an script for recompiling ghdl with prevs commits and run the tests. But it will take a lot of runtime.

What do you think? Which is the best strategy for you? Brute force way?

cmarqu, 2019-02-20:
Let's ping @paebbels, maybe he has an idea.

Paebbels, 2019-02-20:
There have been no time investments to keeping compatibility between GHDL and cocotb, because the cocotb project was dead for so long time. We know that cocotb support is broken.

In some cases that's related to not yet implemented (or buggy?) VPI interface calls. GHDL does not support VHPI, only VPI.

I suggest to create an issue in GHDL and request that GHDL is checking cocotb compatibility again and that it then checks your test case.

benreynwar, 2019-02-21:
The most recent ghdl isn't working with cocotb at all.  I can't even toggle a clk without getting "vpi_get_str: unhandled property".

This error was introduced about a week ago with commit (https://github.com/ghdl/ghdl/commit/4b9a513b53b5a121a1abe5fd93abc8d14e6b7750).

cmarqu, 2019-02-21:
Oh, then let's ping @xnorme (who also uses cocotb).

tgingold, 2019-02-21:
Sorry, but I cannot reproduce the issue.
Using:
```
COCOTB=~/nightly/cocotb make CMD=ghdl PYTHON_BIN=python3.7
```
I get:
```
     -.--ns INFO     cocotb.gpi                                  gpi_embed.c:111  in embed_init_python               Did not detect Python virtual environment. Using system-wide Python interpreter.
     -.--ns INFO     cocotb.gpi                                GpiCommon.cpp:91   in gpi_print_registered_impl       VPI registered
VPI module loaded!
    0.00ns INFO     cocotb.gpi                                  gpi_embed.c:278  in embed_sim_init                  Running on GHDL version 0.1
    0.00ns INFO     cocotb.gpi                                  gpi_embed.c:279  in embed_sim_init                  Python interpreter initialised and cocotb loaded!
    0.00ns INFO     cocotb                                      __init__.py:131  in _initialise_testbench           Running tests with Cocotb v1.1.0 from /Users/gingold/nightly/cocotb
    0.00ns INFO     cocotb                                      __init__.py:147  in _initialise_testbench           Seeding Python random module with 1550778785
    0.00ns INFO     cocotb.regression                         regression.py:209  in tear_down                       Passed 0 tests (0 skipped)
    0.00ns INFO     cocotb.regression                         regression.py:390  in _log_sim_summary                *************************************************************************************
                                                                                                                    **                                 ERRORS : 0                                      **
                                                                                                                    *************************************************************************************
                                                                                                                    **                               SIM TIME : 0.00 NS                                **
                                                                                                                    **                              REAL TIME : 0.00 S                                 **
                                                                                                                    **                        SIM / REAL TIME : 0.00 NS/S                              **
                                                                                                                    *************************************************************************************
                                                                                                                    
    0.00ns INFO     cocotb.regression                         regression.py:218  in tear_down                       Shutting down...
```

xnorme, 2019-02-22:
Trying to come up with a fix in ghdl/ghdl#759. I apologize for the inconvenience. It would be great someone experiencing the issue could give that PR a try before it's merged. 
Ideally we should probably add a minimal testcase for this in GHDL.

eric-wieser, 2019-11-27:
Is this issue still broken? @kammoh, I see that PR was closed without merging - is there a follow-up somewhere?

ktbarrett, 2025-03-27:
AFAIK this has been fixed. `dut.clk` which is presumably a `std_logic` is now discovered by cocotb as a `LogicObject` and assignment should work. Please open a new issue if this is somehow not the case.

closed 2025-03-27 by ktbarrett