cocotb/cocotb #818: VPI: `vpiPort` should be a ModifiableObject?

xnorme, 2019-02-19:
As mentioned in #796 in the current implementation of the VPI interface ([here](https://github.com/potentialventures/cocotb/blob/db2f4e02ae3f788218a2b915f16016b54c52b6eb/lib/vpi/VpiImpl.cpp#L137) and [here](https://github.com/potentialventures/cocotb/blob/db2f4e02ae3f788218a2b915f16016b54c52b6eb/lib/vpi/VpiImpl.cpp#L200)) , `vpiPort`s are made into `HierarchicalObject`s. I believe this is not correct and a `vpiPort` should be considered as a `ModifiableObject` instead. 
I already have a patch for this but I'd like to hear more about the original rational and also #796 to settle.

cmarqu, 2019-12-08:
For reference,  the mapping from ``GPI_MODULE`` to ``HierarchyObject`` is done here:
https://github.com/cocotb/cocotb/blob/db2f4e02ae3f788218a2b915f16016b54c52b6eb/cocotb/handle.py#L748-L751

cmarqu, 2019-12-08:
I agree that a port should not be a ``HierarchyObject`` because this is a subclass of ``RegionObject`` which has the following documentation
"Region objects don't have values, they are effectively scopes or namespaces."
(moved from comments to docstring a short while ago).

cmarqu, 2019-12-08:
Also interesting:
https://github.com/cocotb/cocotb/blob/ddd344d01e1fe93db0786ed026e7ee29d6415ac1/tests/test_cases/test_verilog_access/test_verilog_access.py#L33-L38
and
https://github.com/cocotb/cocotb/blob/ddd344d01e1fe93db0786ed026e7ee29d6415ac1/tests/test_cases/test_vhdl_access/test_vhdl_access.py#L95-L100

garmin-mjames, 2020-09-24:
While investigating #497 / #2075, I found that `vpiPort` objects don't have values themselves and just detail connections.
Using `vpi_handle(vpiLowConn, port_handle)` gets the handle of whatever it's connected to (or NULL for no connection) lower in the hierarchy.
So I think `ModifiableObject` is not the correct mapping for a `vpiPort`.

closed 2020-12-13 by github-actions[bot]