cocotb/cocotb #820: FLI causes stack trace in mixed language simulations

leftink, 2019-02-20:
In mixed language simulations, I have seen issues with the dynamic casting causing stack traces.  It doesn't always happen and can be totally dependent on the type of top one is using.  The problem is, the code made some assumptions about the 'parent' handle when doing the cast and that assumption isn't always holding true, i.e. the parent isn't an FLI handle, causing the cast to fail and hence a stack trace.

I believe we can get rid of the need to do the casting, which will remove the risk of the stack traces.

eric-wieser, 2019-02-24:
Please include the stack trace in this issue, so that someone else can evaluate whether your fix is appropriate. It's hard to review a bug fix without knowing what the bug is.

leftink, 2019-02-24:
@eric-wieser, I am not able to add the stack trace as it was produced at work and we are not allowed to take those type of errors off of our network.  However, I was able to track the problem using GDB and tracked the problem to line 313 of FliImpl.cpp.

```c

        /* If not found, check to see if the name of a generate loop and create a pseudo-region */
        for (rgn = mti_FirstLowerRegion(parent->get_handle<mtiRegionIdT>()); rgn != NULL; rgn = mti_NextRegion(rgn)) {
            if (acc_fetch_fulltype(rgn) == accForGenerate) {
                std::string rgn_name = mti_GetRegionName(static_cast<mtiRegionIdT>(rgn));
                if (rgn_name.compare(0,name.length(),name) == 0) {
                    FliObj *fli_obj = dynamic_cast<FliObj *>(parent);
                    return create_gpi_obj_from_handle(parent->get_handle<HANDLE>(), name, fq_name, fli_obj->get_acc_type(), fli_obj->get_acc_full_type());
                }
            }
        }
```

This is the code that caused the problem.  Since the top was verilog, the VPI interface (incorrectly) produced the parent handle.  When the dynamic_cast() was performed the result returned NULL.  Then the NULL handle was used to get the acc type and the acc fulltype causing the stack trace.

This code assumes the parent is an FLI object, and technically should always be true, but since the VPI interface "found" the parent, the code needs to be modified to not rely on the assumption.l

eric-wieser, 2019-02-25:
Do you know what runtime type `parent` actually is on this line?

leftink, 2019-02-25:
The parent type was VpiObjHdl, since the VPI interface found the "region".  You are correct that the return value should have been checked, but that wouldn't fix the issue, it would only allow for a cleaner failure.

eric-wieser, 2019-02-25:
Are you able to construct a minimal mixed-language test-case that shows the problem?

leftink, 2019-02-25:
Unfortunately I tried, but was unable to.  The design I had the problem with was fairly large and complex.

leftink, 2019-02-25:
@eric-wieser, I have added a design and test to the branch that if you try on the current master branch was able to reproduce the stack trace in my environment with questa.

closed 2025-10-16 by ktbarrett