cocotb/cocotb #824: Python error message report

kimduckhwan, 2019-02-21:
In the test python code, I made some error in the function, which is decorated as coroutine, 

`@cocotb.coroutine
def foo(dut,a, b):
  print("works -1")
  dut.d_in <= BinaryValue(a) # causes error here
  print("works -2")`

when I run cocotb (with VCS), it says

works -1
...    0.00ns INFO     cocotb                                      __init__.py:148  in _initialise_testbench           Seeding Python random module with 1550783275                                                                        
    0.00ns INFO     cocotb.regression                         regression.py:186  in initialise                      Found test test_DP.run_test                                                                                         
    0.00ns INFO     cocotb.regression                         regression.py:319  in execute                         Running test 1/1: run_test                                                                                          
    0.00ns INFO     ..coroutine.run_test.0x2aaabb939290       decorators.py:212  in _advance                        Starting test: "run_test"           
...    
1.03ns ERROR    cocotb.regression                         regression.py:298  in handle_result                   Test Failed: run_test (result was TestError)
    1.03ns ERROR    cocotb.regression                         regression.py:206  in tear_down                       Failed 1 out of 1 tests (0 skipped)
    1.03ns INFO     cocotb.regression                         regression.py:373  in _log_test_summary     

but no error message about where, which line number in python code causes the problem. I need to put print statement to see where is the last line processed in python code.

Is there any option to print error message in detail? like verbose mode?

TrampHRC, 2019-02-25:
use `dut.log.info("<message>")`
dut.log is a SimLog instance I think, defined cocotb/log.py
It seems to implement quit the same features as the **logging** module, which is native python build, so go check that out if you want to go further into debugging.

pwiecha, 2019-02-25:
Or: `log = cocotb.logging.getLogger("cocotb.test")`
and then `log.info("")`

cmarqu, 2019-02-26:
I think the question was not about how to "properly" produce output from the testbench, but about how to get a useful backtrace.
@kimduckhwan please check in ``sim_build/sim.log``, is the output there more complete?

kimduckhwan, 2019-03-08:
Thanks. using log.info() provides more info than just print().

@cmarqu I can't find sim_build/sim.log. Do I need to set something?

`$ ls sim_build/
csrc  pli.tab  simv  simv.daidir  vc_hdrs.h
`

cmarqu, 2019-03-10:
@kimduckhwan Sorry, apparently cocotb doesn't write the log into that file for VCS - I opened #838 for that. I'm not sure where the logfile actually is then.
What you may try is to add ``-l $(SIM_BUILD)/sim.log``to either line 70 or 76 in https://github.com/potentialventures/cocotb/blob/master/cocotb/share/makefiles/simulators/Makefile.vcs.

eric-wieser, 2019-12-02:
> Is there any option to print error message in detail? like verbose mode?

This should happen by default now, as part of #988 or perhaps #1100 or #1091. Feel free to comment if that's not the case.

closed 2019-12-02 by eric-wieser