cocotb/cocotb #829: Standard python exception handling and reporting

jeinstei, 2019-02-28:
I'm finding that python exceptions will fail a simulation with the message below, but not actually perform a traceback or report the exception.

`#     0.00ns ERROR    cocotb.scheduler                            __init__.py:190  in _sim_event                      Failing test at simulator request before test run completion: Simulator shutdown prematurely`

My call graph is: cocotb.test > yield cocotb.coroutine > setup_function

I've tried raising RuntimeError, StandardError, ValueError, and NotImplementedError, and TestError, and I get the same results.

eric-wieser, 2019-03-19:
Can you paste the full stack trace? What version of cocotb?

jeinstei, 2019-03-19:
This was actually related to a previous ticket that I close.d I somehow was registering objects multiple times which was breaking things. Let me close this one out and if I still see it will reopen with more information.

There was actually no stack trace, which was the entire problem in the first place. Things might have changed now though.

closed 2019-03-19 by jeinstei