cocotb/cocotb #830: Monitor/BusMonitor kwargs

jeinstei, 2019-02-28:
Currently BusMonitor and Monitor don't accept additional kwargs, meaning that you can't as easily instantiate and pass down arguments with a basic **kwargs. Adding **kwargs to the end of the definition would make life easier.

eric-wieser, 2019-03-16:
Can you give an example of what you want to be able to do here?

jeinstei, 2019-03-16:
This is more a consistency issue as the drivers base classes pass kwargs in while the monitors don't. As long as thiggs are consistent so that writing classes is the same style, it makes things look better in higher level code

eric-wieser, 2019-03-16:
> the drivers base classes pass kwargs

Can you link to the bit of code you're talking about?

eric-wieser, 2019-03-16:
Ah, I see:
```
    def __init__(self, entity, name, clock, **kwargs):
        self.log = SimLog("cocotb.%s.%s" % (entity._name, name))
        Driver.__init__(self)
        self.entity = entity
        self.clock = clock
        index = kwargs.get("array_idx")
        self.bus = Bus(self.entity, name, self._signals,
                       self._optional_signals, array_idx=index)

        # Give this instance a unique name
        self.name = name if index is None else "%s_%d" % (name, index)
```

I would consider this a bug - it should be spelt

```diff
 def __init__(self, entity, name, clock, **kwargs):
         Driver.__init__(self)
         self.entity = entity
         self.clock = clock
-        index = kwargs.get("array_idx")
+        index = kwargs.pop("array_idx")
+        if kwargs:
+            bad_arg = next(iter(kwargs))
+            raise TypeError('__init__() got an unexpected keyword argument {!r}'.format(bad_arg))
         self.bus = Bus(self.entity, name, self._signals,
                        self._optional_signals, array_idx=index)
```

eric-wieser, 2019-03-16:
Actually, [I caught this bug in review](https://github.com/potentialventures/cocotb/pull/708#discussion_r241969404), but the review was so unwieldy it got missed.

imphil, 2019-03-17:
@eric-wieser thanks for the fix in https://github.com/potentialventures/cocotb/pull/860, and sorry your review comment got missed in #708.

imphil, 2019-03-17:
@jeinstei I hope the solution @eric-wieser has found works for you. Please re-open if there are remaining blockers for your use case.

PR #860 (2019-03-17): Reject extra kwargs in BusDriver.__init__
Silently discarded misspelt kwargs is a recipe for disaster.

This fixes #830 by rejecting the kwargs in BusDriver rather than accepting them in BusMonitor

This also fixes the same problem in wavedrom
commit 5b4737b3 (2019-03-17): Reject extra kwargs in BusDriver.__init__

Silently discarded misspelt kwargs is a recipe for disaster.

This fixes #830 by rejecting the kwargs in BusDriver rather than accepting them in BusMonitor

This also fixes the same problem in wavedrom
closed 2019-03-17 by imphil