cocotb/cocotb #832: Prime/unprime on RisingEdge and deregistered cbValueChange

jeinstei, 2019-02-28:
I'm using a RisingEdge(clock) that seems to de-register itself unless I unprime it and then re-prime it. I was wondering if creating a trigger causes any sort of odd interaction with vpi that might be a known issue, but not reported here? I'm seeing errors on priming from the Vpi c++ file, not in Python, which is making me wonder about call graph issues.

```
Attempt to prime an already primed trigger for cbValueChange!
We seem to already be registered, deregistering cbValueChange!
```

jeinstei, 2019-02-28:
Related, this is in a class that is a child of ValidatedBusDriver. I have another ValidatedBusDriver that was used previously. Having multiple BusDrivers shouldn't be an issue, but I thought I should point this out.

garmin-mjames, 2019-02-28:
Are you trying to use an object of your ValidatedBusDriver subclass across multiple tests?

All coroutines and triggers are killed/unprimed at the end of a cocotb test (see Scheduler.cleanup()).

The scheduler should see the trigger already primed and not call down to the VPI code.

jeinstei, 2019-02-28:
Okay-- this clarified what was going on. I was replicating objects, basically. Coroutines and triggers are unprimed, but it seems like it might be important to keep a global container for testbench objects to only create the major drivers/monitors once. I think I was creating multiple objects that were drivers somehow causing overwrites somewhere. I think it is still a bug somewhere, but different.

jeinstei, 2019-03-18:
Closing for now as not blocking anything and not reporting the actual bug itself. If it shows up again I'll look deeper in to it.

closed 2019-03-18 by jeinstei