cocotb/cocotb #833: Python 3.7, StopIteration

mox-mox, 2019-03-02:
I've been trying out cocotb. When trying to make the cocotb with python 3.7, the tests fail on
tests/test_cases/test_iteration_verilog/test_iteration_es.py, line 50 with the following traceback:
```
128000000000.00ns ERROR    ..ecursive_discovery.0x7f4a391e8940           result.py:52   in raise_error                     Send raised exception:
                                                                                                                           Traceback (most recent call last):
                                                                                                                             File "/home/mox/projects/iverilog/cocotb/cocotb/cocotb/handle.py", line 509, in __iter__
                                                                                                                               raise StopIteration
                                                                                                                           StopIteration

                                                                                                                           The above exception was the direct cause of the following exception:

                                                                                                                           Traceback (most recent call last):
                                                                                                                             File "/home/mox/projects/iverilog/cocotb/cocotb/cocotb/decorators.py", line 218, in _advance
                                                                                                                               return outcome.send(self._coro)
                                                                                                                             File "/home/mox/projects/iverilog/cocotb/cocotb/cocotb/outcomes.py", line 37, in send
                                                                                                                               return gen.send(self.value)
                                                                                                                             File "/home/mox/projects/iverilog/cocotb/cocotb/tests/test_cases/test_iteration_verilog/test_iteration_es.py", line 66, in recursive_discovery
                                                                                                                               total = dump_all_the_things(dut)
                                                                                                                             File "/home/mox/projects/iverilog/cocotb/cocotb/tests/test_cases/test_iteration_verilog/test_iteration_es.py", line 62, in dump_all_the_things
                                                                                                                               count += dump_all_the_things(thing)
                                                                                                                             File "/home/mox/projects/iverilog/cocotb/cocotb/tests/test_cases/test_iteration_verilog/test_iteration_es.py", line 57, in dump_all_the_things
                                                                                                                               for thing in parent:
                                                                                                                           RuntimeError: generator raised StopIteration
```
Googling pointed me to [this stackoverflow answer](https://stackoverflow.com/a/51701040) which references [PEP 479 -- Change StopIteration handling inside generators](https://www.python.org/dev/peps/pep-0479/). I'm not very deep into python, but it seems to me the different handling of StopIteration described in PEP 479 is the cause of the issue.

cmarqu, 2019-03-11:
Related?: #649

imphil, 2019-03-17:
Thanks for reporting this bug. This should be fixed by #849. Please re-open if the issue still exists on the current master.

closed 2019-03-17 by imphil