cocotb/cocotb #834: Mutliple Test Failures for Questa and Riviera

leftink, 2019-03-06:
# Description
While checking that other fixes I am working on didn't break anything, I noticed several test failures when running regressions.  Both QuestaSIM and Riviera-Pro have failures.  See below for commands and result summaries.  I have attached the logs of the failures.

This issue may need to be split into multiple issues.

# Environment
* Linux 64-bit (Redhad 6)
* Anaconda 4.0.0 (Python 3.6)
* QuestaSIM (10.5c)
* Riviera-Pro (2018.02)
* COCOTB (8cb1e9e)

#  QuestaSIM Default
`make SIM=questa SIM_ARGS=-novopt`
## QuestaSIM Default Results
```
Failure in testsuite: 'all' classname: 'test_cocotb_array' testcase: 'test_in_2d_arr_unpacked' with parameters 'all'
Failure in testsuite: 'all' classname: 'test_cocotb_array' testcase: 'test_in_arr_packed_unpacked' with parameters 'all'
Failure in testsuite: 'all' classname: 'test_cocotb_array' testcase: 'test_in_vect_packed_packed_unpacked' with parameters 'all'
Ran a total of 1 TestSuites and 167 TestCases
```
Results: [questa_default_results.zip](https://github.com/potentialventures/cocotb/files/2938176/questa_results_1.zip)

#  QuestaSIM VHDL
`make SIM=questa SIM_ARGS=-novopt TOPLEVEL_LANG=vhdl`
## QuestaSIM VHDL Results
```
make[3]: *** [results.xml] Error 211
make[3]: Leaving directory `/scratch/z7023396/git_wa/rms-cdl-reuse/cocotb/examples/axi_lite_slave/tests'
make[2]: *** [sim] Error 2
make[2]: Target `all' not remade because of errors.
make[2]: Leaving directory `/scratch/z7023396/git_wa/rms-cdl-reuse/cocotb/examples/axi_lite_slave/tests'
make[1]: *** [axi_lite_slave/tests] Error 2
make[1]: Target `all' not remade because of errors.
make[1]: Leaving directory `/scratch/z7023396/git_wa/rms-cdl-reuse/cocotb/examples'
make: *** [do_tests] Error 2

--- FORCED TO MANUALLY RUN COMBINE_RESULTS ---
./bin/combine_results
Failure in testsuite: 'all' classname: 'issue_768' testcase: 'do_test' with parameters 'all'
Failure in testsuite: 'all' classname: 'test_cocotb' testcase: 'test_readwrite_in_readonly' with parameters 'all'
Failure in testsuite: 'all' classname: 'test_discovery' testcase: 'access_internal_register_array' with parameters 'all'
Ran a total of 1 TestSuites and 134 TestCases
```
Results: [questa_vhdl_results.zip](https://github.com/potentialventures/cocotb/files/2938187/questa_results_2.zip)

# RivieraPro Default
`make SIM=aldec SIM_ARGS=-novopt`
## RivieraPro Default Results
```./bin/combine_results.py
Ran a total of 1 TestSuites and 167 TestCases
```
Results: No File (All Passed)

# RivieraPro VHDL
`make SIM=aldec SIM_ARGS=-novopt TOPLEVEL_LANG=vhdl`
## RivieraPro VHDL Results 
```
./bin/combine_results.py
Failure in testsuite: 'all' classname: 'test_adder' testcase: 'adder_basic_test' with parameters 'all'
Failure in testsuite: 'all' classname: 'test_adder' testcase: 'adder_randomised_test' with parameters 'all'
Failure in testsuite: 'all' classname: 'test_endian_swapper' testcase: 'run_test_002' with parameters 'all'
Failure in testsuite: 'all' classname: 'test_endian_swapper' testcase: 'run_test_003' with parameters 'all'
Failure in testsuite: 'all' classname: 'test_endian_swapper' testcase: 'run_test_004' with parameters 'all'
Failure in testsuite: 'all' classname: 'test_endian_swapper' testcase: 'run_test_005' with parameters 'all'
Failure in testsuite: 'all' classname: 'test_endian_swapper' testcase: 'run_test_006' with parameters 'all'
Failure in testsuite: 'all' classname: 'test_endian_swapper' testcase: 'run_test_007' with parameters 'all'
Failure in testsuite: 'all' classname: 'test_endian_swapper' testcase: 'run_test_008' with parameters 'all'
Failure in testsuite: 'all' classname: 'test_endian_swapper' testcase: 'run_test_009' with parameters 'all'
Failure in testsuite: 'all' classname: 'test_endian_swapper' testcase: 'run_test_010' with parameters 'all'
Failure in testsuite: 'all' classname: 'test_endian_swapper' testcase: 'run_test_011' with parameters 'all'
Failure in testsuite: 'all' classname: 'test_endian_swapper' testcase: 'run_test_012' with parameters 'all'
Failure in testsuite: 'all' classname: 'test_endian_swapper' testcase: 'run_test_013' with parameters 'all'
Failure in testsuite: 'all' classname: 'test_endian_swapper' testcase: 'run_test_014' with parameters 'all'
Failure in testsuite: 'all' classname: 'test_endian_swapper' testcase: 'run_test_015' with parameters 'all'
Failure in testsuite: 'all' classname: 'test_endian_swapper' testcase: 'run_test_016' with parameters 'all'
Failure in testsuite: 'all' classname: 'test_endian_swapper' testcase: 'run_test_018' with parameters 'all'
Failure in testsuite: 'all' classname: 'test_endian_swapper' testcase: 'run_test_019' with parameters 'all'
Failure in testsuite: 'all' classname: 'test_endian_swapper' testcase: 'run_test_020' with parameters 'all'
Failure in testsuite: 'all' classname: 'test_endian_swapper' testcase: 'run_test_021' with parameters 'all'
Failure in testsuite: 'all' classname: 'test_endian_swapper' testcase: 'run_test_022' with parameters 'all'
Failure in testsuite: 'all' classname: 'test_endian_swapper' testcase: 'run_test_023' with parameters 'all'
Failure in testsuite: 'all' classname: 'test_endian_swapper' testcase: 'run_test_024' with parameters 'all'
Failure in testsuite: 'all' classname: 'test_endian_swapper' testcase: 'run_test_025' with parameters 'all'
Failure in testsuite: 'all' classname: 'test_endian_swapper' testcase: 'run_test_026' with parameters 'all'
Failure in testsuite: 'all' classname: 'test_endian_swapper' testcase: 'run_test_027' with parameters 'all'
Failure in testsuite: 'all' classname: 'test_endian_swapper' testcase: 'run_test_028' with parameters 'all'
Failure in testsuite: 'all' classname: 'test_endian_swapper' testcase: 'run_test_029' with parameters 'all'
Failure in testsuite: 'all' classname: 'test_endian_swapper' testcase: 'run_test_030' with parameters 'all'
Failure in testsuite: 'all' classname: 'test_endian_swapper' testcase: 'run_test_031' with parameters 'all'
Failure in testsuite: 'all' classname: 'test_endian_swapper' testcase: 'run_test_032' with parameters 'all'
Failure in testsuite: 'all' classname: 'issue_768' testcase: 'do_test' with parameters 'all'
Failure in testsuite: 'all' classname: 'test_cocotb' testcase: 'test_clock_with_units' with parameters 'all'
Failure in testsuite: 'all' classname: 'test_cocotb' testcase: 'test_edge_count' with parameters 'all'
Failure in testsuite: 'all' classname: 'test_cocotb' testcase: 'test_falling_edge' with parameters 'all'
Failure in testsuite: 'all' classname: 'test_cocotb' testcase: 'test_rising_edge' with parameters 'all'
Failure in testsuite: 'all' classname: 'test_discovery' testcase: 'access_internal_register_array' with parameters 'all'
Ran a total of 1 TestSuites and 134 TestCases
```
Results: [rivierapro_vhdl_results.zip](https://github.com/potentialventures/cocotb/files/2938203/riviera_results_2.zip)

ktbarrett, 2019-12-19:
Current (17ce4e51da43b6d2d1b5834d629374ed5535a8d8) results for Questa.

* RHEL 7
* GCC 4.8.5
* Python 3.6.1
* Questa 10.5c

### Verilog
```shell
 $ make SIM=questa SIM_ARGS=-novopt
...
 $ ./bin/combine_results.py
Failure in testsuite: 'all' classname: 'test_closedown' testcase: 'test_failure_from_system_task' with parameters 'all'
Failure in testsuite: 'all' classname: 'test_cocotb_array' testcase: 'test_in_2d_arr_unpacked' with parameters 'all'
Failure in testsuite: 'all' classname: 'test_cocotb_array' testcase: 'test_in_arr_packed_unpacked' with parameters 'all'
Failure in testsuite: 'all' classname: 'test_cocotb_array' testcase: 'test_in_vect_packed_packed_unpacked' with parameters 'all'
Failure in testsuite: 'all' classname: 'plusargs' testcase: 'plusargs_test' with parameters 'all'
Failure in testsuite: 'all' classname: 'test_discovery' testcase: 'access_var_string_verilog' with parameters 'all'
Ran a total of 1 TestSuites and 159 TestCases
```

### VHDL
```shell
 $ make SIM=questa SIM_ARGS=-novopt TOPLEVEL_LANG=vhdl
...
 $ ./bin/combine_results.py 
Failure in testsuite: 'all' classname: 'test_array' testcase: 'test_read_write' with parameters 'all'
Failure in testsuite: 'all' classname: 'test_closedown' testcase: 'test_failure_from_system_task' with parameters 'all'
Failure in testsuite: 'all' classname: 'test_cocotb_array' testcase: 'test_in_2d_arr_unpacked' with parameters 'all'
Failure in testsuite: 'all' classname: 'test_cocotb_array' testcase: 'test_in_arr_packed_unpacked' with parameters 'all'
Failure in testsuite: 'all' classname: 'test_cocotb_array' testcase: 'test_in_vect_packed_packed_unpacked' with parameters 'all'
Failure in testsuite: 'all' classname: 'test_cocotb' testcase: 'test_readwrite_in_readonly' with parameters 'all'
Failure in testsuite: 'all' classname: 'plusargs' testcase: 'plusargs_test' with parameters 'all'
Failure in testsuite: 'all' classname: 'test_discovery' testcase: 'access_internal_register_array' with parameters 'all'
Failure in testsuite: 'all' classname: 'issue_957' testcase: 'test1' with parameters 'all'
Failure in testsuite: 'all' classname: 'issue_957' testcase: 'test1' with parameters 'all'
Ran a total of 1 TestSuites and 161 TestCases
```

### Issues
* plusargs fails because of issues with the makefile #1244.
* issue_957 fails because VHDL's example module is out of date (trying to reference `dut.and_output` that exists in SV tb top, but not in VHDL tb top).
* test_closedown fails because the test `expect_fail=True`, although `SimFailure` is treated as an error (this should be `expect_error=SimFailure`).
* access_var_string_verilog fail because `int VpiSignalObjHdl::set_signal_value(std::string &value)
` sets the format to `vpiBinStrVal` instead of the proper `vpiStringVal`.
* test_array fails because the test writes an uppercase 'E' to the string port at location two, turning `Testing` into `TEsting`, but it checks later against `Testing` and fails.
* test_readwrite_in_readonly fails because we expect a failure, but it passes (investigate, why do we expect it to fail?).
* test_cocotb_array needs more investigation.

cmarqu, 2019-12-21:
> * plusargs fails because of issues with the makefile #1244.

In the VHDL version, it should also fail because of #1242.

ktbarrett, 2021-06-24:
Continuing in #2579.

closed 2021-06-24 by ktbarrett