cocotb/cocotb #835: Quick Reference Card/Cheat Sheet

cmarqu, 2019-03-07:
We should have a one-page quick reference card in the style of http://docs.sphinxdocs.com/en/latest/cheatsheet.html and http://www.tcnj.edu/~hernande/r/VHDL_QRC__01.pdf etc.

Topics:
* how to assign/read/print value in different formats
* yield
* fork/join/kill a coroutine
* do checks
* finish test
* ...

(please suggest more below).

The card should show which imports are needed for each item.

pwiecha, 2019-03-08:
I think a good start would be to reuse what Luke has done for Orconf 2018 (with his permission ofc).
https://github.com/lukedarnell/cocotb/tree/master/tests/test_cases/orconf2018

ktbarrett, 2024-05-18:
@cmarqu this closed right?

closed 2024-05-18 by cmarqu