cocotb/cocotb #839: Result of a yield of a list of triggers is not always in the list

jwprice100, 2019-03-11:
This issue occurs when dealing with a series of forking coroutines which read signals driven by the design under test and cocotb. Under the right conditions the result returned by yielding a list of triggers is not in the list of triggers.  Another related issue is that under these same conditions the erroneous trigger will fire over and over. Eventually simulation resumes but not for a significant time. 

**Context:**
CentOS based system running Python 3.6
Using VHDL2008
Simulator: Riviera Pro 2018.10
Added -fgnu89-inline switch to gcc_args in order for CoCoTb to compile on my system

Please note I am new to CoCoTb and some of what I first believed to be bugs were in fact user error. This may be as well but I have not been able to determine why. I've attached a simple example that reproduces the issue.

[simple_coco_testcase.zip](https://github.com/potentialventures/cocotb/files/2950395/simple_coco_testcase.zip)

jwprice100, 2019-03-11:
Digging more into this issue, I believe I know what one of the issues I am seeing is. 

Consider the case of two coroutines A and B. A has a single trigger (say RisingEgde of clock) that it yields two indefintely. Coroutine B also a list of triggers it yields to, then at a later point yields to a different list. That second list includes the trigger that coroutine A uses. 

When a trigger fires, the scheduler considers all the other triggers a scheduled coroutine is associated with that did not fire. For a particular pending trigger it checks if the set of coroutines that will be scheduled is a superset of the routines associated with the trigger. If so, it unprimes the trigger and removes it. This makes sense as if all coroutines that are waiting on that pending trigger awoke due a different trigger they are not relevant.

However consider the case where a pending trigger which did not fire and is attached to a different coroutine that is not being scheduled to run as in my example.

The code which handles this is given below:
```python
          for pending in other_triggers:
                # every coroutine waiting on this trigger is already being woken
                if scheduling_set.issuperset(self._trigger2coros[pending]):
                    if pending.primed:
                        pending.unprime()
                    del self._trigger2coros[pending]
```

My hacky workaround is as follows:
```python
            for pending in other_triggers:
                # every coroutine waiting on this trigger is already being woken
                if scheduling_set.issuperset(self._trigger2coros[pending]):
                    if pending.primed:
                        pending.unprime()
                    del self._trigger2coros[pending]
                else:
                     #For each coroutine
                     for coro in scheduling_set:
                        #If that coroutine is associated with a pending trigger
                        if coro in self._trigger2coros[pending]:
                           #Remove that coroutine
                           index = self._trigger2coros[pending].index(coro)
                           self._trigger2coros[pending].pop(index)
```

Can anyone with more experience with the scheduler confirm my thoughts?

eric-wieser, 2019-03-19:
Here's a minimal repro:
```python
@cocotb.test()
def test_yield_list_stale(dut):
    """ Test that a trigger yielded as part of a list can't cause a spurious wakeup """
    # gh-843
    events = [Event() for i in range(3)]

    waiters = [e.wait() for e in events]

    @cocotb.coroutine
    def wait_for_lists():
        ret_i = waiters.index((yield [waiters[0], waiters[1]]))
        assert ret_i == 0, "Expected event 0 to fire, not {}".format(ret_i)

        ret_i = waiters.index((yield [waiters[2]]))
        assert ret_i == 0, "Expected event 2 to fire, not {}".format(ret_i)

    @cocotb.coroutine
    def wait_for_e1():
        """ wait on the event that didn't wake `wait_for_lists` """
        ret_i = waiters.index((yield waiters[1]))
        assert ret_i == 1, "Expected event 1 to fire, not {}".format(ret_i)

    @cocotb.coroutine
    def fire_events():
        """ fire the events in order """
        for e in events:
            yield Timer(1)
            e.set()

    fire_task = cocotb.fork(fire_events())
    e1_task = cocotb.fork(wait_for_e1())
    yield wait_for_lists()

    # make sure the other tasks finish
    yield fire_task.join()
    yield e2_task.join()
```

eric-wieser, 2019-04-05:
I think this can be closed by #870

eric-wieser, 2019-05-06:
@imphil: close?

imphil, 2019-05-06:
@jwprice100 this issue should be resolved in the current master of cocotb. Please reopen this issue if that's not the case.

closed 2019-05-06 by imphil