cocotb/cocotb #840: Simulation time precision limitations

pwiecha, 2019-03-11:
How should one set a simulator time precision?

Let's take questa for example, `cocotb/makefiles/simulators/Makefile.questa` defines timescale as 
`-timescale 1ns/100ps`
So no clock with period < 100 ps and only multiples of 100 ps possible by default.

To get around this I've added `-timescale 1ns/1ps` to the COMPILE_ARGS var. But it's ugly, questa basically has 2 timescale directives now and overwrites the default one.

In case of Icarus, one can do `EXTRA_ARGS += -c $(PWD)/iverilog_precision` (where iverilog_precision is a file containing timescale directive in format `+timescale+1ns/1ps`) Notice the different timescale format.

Don't know how to do this in xrun/ius yet.

I don't think that the end user should do these kind of hacks in Makefile. Let's discuss a possible implementation of the uniform interface for setting this across different simulators.

jeinstei, 2019-03-25:
Are you trying to override the entire simulation, or just the default? If it is defined in a package or file in the project, that's going to change things up.

cmarqu, 2021-08-29:
This is fixed since quite some time now.
Also related: https://github.com/cocotb/cocotb/pull/2685

closed 2021-08-29 by cmarqu