cocotb/cocotb #841: Scheduler issue when coroutine waiting on list of triggers

garmin-mjames, 2019-03-11:
There is a bug in the scheduler dealing with unpriming triggers correctly when a coroutine in waiting on a list of triggers, and one of those triggers fires, but other coroutine(s) are waiting on a non-fired trigger in the list.

The scenario is as follows:
Two coroutines exist, with coro_A waiting on a list of triggers [RisingEdge(sig1), RisingEdge(sig2)], and coro_B waiting on one of the triggers in coco_A's list RisingEdge(sig1).
If one of the **other** triggers in coro_A's list fires, RisingEdge(sig2), the scheduler does not unprime RisingEdge(sig1) because coro_B is waiting on it. This is correct behavior.
The bug is that the scheduler also does not update the list of coroutines that are waiting on the trigger RisingEdge(sig1). The scheduler should remove coro_A from this list, as it is no longer waiting on it.

This will cause coro_A to be called when RisingEdge(sig1) does happen in the future.

```
@cocotb.coroutine
def coro_A(dut):
    while True:
        yield [RisingEdge(dut.sig1), RisingEdge(dut.sig2)]

@cocotb.coroutine
def coro_B(dut):
    while True:
        yield RisingEdge(sig1)

@cocotb.test
def test1(dut):
    cocotb.fork(coro_A(dut))
    cocotb.fork(coro_B(dut))

    yield Timer(1)
    dut.sig2.value = '1'
```

This may be related to issue #839 

I have a fix for this and will create a Pull Request.

eric-wieser, 2019-03-21:
@garmin-mjames: I don't really understand your test, so I can't tell if this is a duplicate of #839. Can you test this on master and see if it can be closed?

garmin-mjames, 2019-03-21:
The test is designed for two coroutines to yield on the same trigger, but with one of the two coroutines also waiting on an additional trigger. The list of pending coroutines for the additional trigger was not being cleaned up correctly.

It is a duplicate of #839 because it is caused by the same issue in the scheduler.
Now that #870 is merged I'll test it.

garmin-mjames, 2019-03-28:
I tested this on master and #870 fixed this.

closed 2019-03-28 by garmin-mjames