cocotb/cocotb #852: Coroutine hangs when yielding a Combine Trigger of coroutine joins

leftink, 2019-03-13:
In a coroutine, I fork multiple coroutines and then want to block until all coroutines are complete.  I tried using the Combine trigger, but it never fired so the coroutine hangs indefinitely.

A simple example of the problem:

```python
@coroutine
def do_something(hdl, delay):
    hdl <= 1
    yield Timer(delay)
    hdl <= 0

@coroutine
def do multiple_things(hdl):
    crs = [fork(do_something(hdl, dly)) for dly in [10, 30, 20]]

    yield Combine(*(cr.join() for cr in crs))
```

The issue is that a Join() trigger doesn't call the callback provided during prime() unless the coroutine has already completed.  As a result the Combine trigger never gets notified of each coroutine completing.

Don't have a solution to the problem as it will require changes to scheduler and possibly the Join trigger.

leftink, 2019-03-17:
@imphil and @eric-wieser, I would like to start a little discussion on this issue as I know Eric has been doing a lot of work on triggers.

There seems to be a lot of special handling of the Join trigger in the scheduler.  The way the Join trigger is written/handled is what prevents the Combine trigger from functioning correctly.  If we really think about what we are trying to accomplish with the `join()` functionality, the Join trigger could be replaced with an Event trigger where the join() function would invoke wait() on the Event trigger and before exiting the coroutine would invoke set() on the trigger waking any coroutines waiting on it and removing that functionality from the scheduler.

On a similar note, the scheduler is doing special handling of a list of triggers/coroutines.  Would it make more sense to create a trigger that is like the Combine trigger, to properly manage the trigger handling?  The scheduler could still accept a list and then create the new trigger class.  To parallel python we could have two triggers, `Any` and `All`.  The `All` trigger would just be a subclass of the Combine trigger to maintain backward compatibility.  Maybe we could even support  yielding a `tuple` to be an `All`.

A thought just occurred, that we may have a bigger issue with the `Combine` trigger.  The `Combine` trigger primes the children trigger with a different callback then `Scheduler.react()`, but for example if an Edge Trigger is already primed by another coroutine and code is written to include an Edge trigger for the same signal with the Combine trigger, the Combine trigger will never fire because the Edge trigger was already primed with the callback to the scheduler.  Given that maybe Combine will need to be handled in a similar way as the list, i.e. in the Scheduler.

eric-wieser, 2019-03-17:
You can fix this as of just now by writing:
```
class Combine:
    def __init__(self, triggers):
        self._triggers = triggers
    def __await__(self):
        t = list(self._triggers)
        while t:
            fired = yield t
            t.remove(fired)
```

Unfortunately this only works with the async/await syntax

eric-wieser, 2019-03-21:
This is now fixed in master

closed 2019-03-21 by imphil