cocotb/cocotb #854: Trigger/Coroutine Clean-up After test has issues

leftink, 2019-03-14:
I had a module with three tests, the first test ran fine, the second test would hang (timeout, in my case), then the 3rd test would run fine.

Digging in found the second test was waiting on the ReadOnly trigger to fire.

I haven't had time to figure out how this was happening, but what I was able to see by adding debug logs was that after `scheduler.default_scheduling_algorithm()` was called with the _terminate, i.e. previous test had ended, there is some clean-up that occurs to get ready for the next step, there were coroutines still running that were forked and yielding things back.  This caused the scheduler to be in an invalid state at the start of the next test.  I also saw that during the clean-up, the scheduler didn't believe the ReadOnly trigger was primed, so it didn't unprime it, and then I saw the readonly callback fire before the next test started.

As a quick workaround I was able to get past my problem by modifying one function in the scheduler to protect it from executing if the scheduler is terminating:
```python

    def _coroutine_yielded(self, coro, triggers):
        """Prime the triggers and update our internal mappings."""
        if self._mode == Scheduler._MODE_TERM:
            return

        ...
```

While this solved my problem, it got me really questioning if the Trigger processing is correct in the schedulre, so I believe more investigation is required prior to a fix.

One thought, not fully investigated, why does the scheduler do so much checking the state of triggers and controlling the state of triggers.  The scheduler checks if a trigger is primed before priming or unpriming, why not just push that to the trigger and have the scheduler always call prime/unprime.  Further, it seems dangerous given all the branches in the scheduler now, to have the scheduler be responsible for calling unprime on a trigger that fired.  Why not have the trigger set the prime state appropriately when it fires?

leftink, 2019-03-17:
@imphil and @eric-wieser, I would like to start a little discussion on this issue as I know Eric has been doing a lot of work on triggers.  Thoughts would be appreciated.

eric-wieser, 2019-03-17:
>  I also saw that during the clean-up, the scheduler didn't believe the ReadOnly trigger was primed, so it didn't unprime it

This is something I spent a few days looking at, and is possibly related to #759

eric-wieser, 2019-05-06:
@leftink: ping

ktbarrett, 2025-02-17:
This is fixed.

closed 2025-02-17 by ktbarrett