cocotb/cocotb #855: XGMII with Avalon interface

bth105, 2019-03-14:
So im new to cocotb. Im trying to use the XGMII on the input interface and the Avalon Packet bus interface on the output. I am using the XGMII monitor to produce the expected output. However i get a transaction mismatch in the scoreboard.
```
              Expected:
                                                                                                                    '\xff\xff\xff\xff\xff\xff\x9c\xb6\xd0\xbf\xb6w\x90\x00\x85\x83\xdc\xf6\x02\x86\xfe\xb2\r\xa6\x80c\xaa\xcc_\xf5\x06w%\x97T\xd6}\x85l\x86S\xf3\xfa\xf6b\xd3\x92Z"\xb3W\xce\xe2\xd7\x97p\xaa\xd06\x14\x83\xdc"\x1d\x82\xdf\xf8cb\xbe\x9c\xe9\x86`\xa8\xd6\x0ePo\x87\xf6\x1e\xaa\xe7P\x1e\xf5\xe4\x80\x0f\x97"\x8e\x82\n\xd4\xea\x03\xc4.-\xb9\xd02\xcfv\xba'
    0.12ns INFO     ..tb.scoreboard.X       scoreboard.py:156  in compare                         Received:
                                                                                                                    '\xff\xff\xff\xff\xff\xff\x9c\xb6'
    0.12ns WARNING  ..tb.scoreboard.X       scoreboard.py:163  in compare                        
```                                                           
I think the XGMII monitor is producing the whole packet as the transaction and the Avalon pkt interface is producing one words width. But i could be mistaken. Have people used the XGMII driver/monitor and Avalon Packet bus in this configuration before.

cmarqu, 2019-03-14:
I haven't used either of them, and I'm just guessing wildly, but has it maybe to do with packetised vs. non-packetised Avalon?: https://cocotb.readthedocs.io/en/latest/library_reference.html#id1

bth105, 2019-03-14:
Oh i feel stupid, your right i had for some reason not being using the avalon pkt interface, even though i thought i was. cheers

closed 2019-03-14 by bth105