cocotb/cocotb #856: Crash when using a SystemVerilog enum in iverilog

mox-mox, 2019-03-14:
When using an enum inside the top-level module, iverilog will crash when trying to access any signal. When I try similar code with plain iverilog, it will not crash.
Using an enum inside a nested module seems to work.

Test code:
`enum.sv`:
```
`default_nettype none
`timescale 1ns/1ns

	typedef enum reg[3:0]
	{
		INVALID  = 4'b0000,
		ONE      = 4'b0001,
		TWO      = 4'b0011,
		THREE    = 4'b0111,
		FOUR     = 4'b1111
	} thermometer;

module example(input wire clk);

	//   |  Uncomment either this...
	//   v
	//typedef enum reg
	//{
	//	FOO = 1'b0,
	//	BAR = 1'b1
	//} something_else;

	//   |  ... or that and the
	//   v   simulator segfaults.
	//thermometer foo;

	always @(posedge clk)
	begin
		$display("Got a rising clock edge");
	end

endmodule
```

`test_enum.py`
```
import cocotb
from cocotb.triggers import Timer

@cocotb.test()
def test_enum(dut):
	"""
	Try enum variables
	"""
	dut._log.info("Running test!")
	for cycle in range(10):
		dut._log.info("X")
		dut.clk = 0
		dut._log.info("Y")
		yield Timer(1000)
		dut.clk = 1
		yield Timer(1000)
	dut._log.info("Running test!")
```
and the `Makefile`:
```
export USER_DIR:=$(shell pwd)

include $(COCOTB)/makefiles/Makefile.inc
include $(COCOTB)/makefiles/Makefile.sim

VERILOG_SOURCES = enum.sv
TOPLEVEL = example
MODULE=test_enum
```

Could you give me some pointers on how to further debug the problem?

Cheers,
mox

cmarqu, 2019-03-15:
You might try getting a backtrace following some instructions like https://jvns.ca/blog/2018/04/28/debugging-a-segfault-on-linux/.
The output may only be useful to iverilog developers though.

Oh, and maybe you are not at the latest iverilog version but could try it?

imphil, 2019-03-17:
This is a bug in iverilog, possibly in their VPI support. Please follow the guide at http://iverilog.icarus.com/support/bugs to report it there, and re-open this bug if additional information from cocotb is required.

closed 2019-03-17 by imphil

mox-mox, 2019-03-21:
Ok, I finally got around to opening a [ticket](https://github.com/steveicarus/iverilog/issues/235) in their github repo. Let's see what happens.

mox-mox, 2019-03-24:
Its fixed upstream :)