cocotb/cocotb #862: Wishbone bus driver and monitor

imphil, 2019-03-17:
In order to keep the cocotb code base maintainable we're looking for ways to keep the core small (and "core"), and move bus drivers and similar functionality into separate modules. Now with cocotb being a Python package we expect this to be much easier, as requirements can be specified in a well-known way and pip will install them.

The following PRs contain work-in-progress code which can serve as a starting point for an Wishbone bus extension:
- #335
- #439
- #454

mkreider, 2019-11-25:
Wow. I've handed in the original code for a WB bus module in 2015 and it wasn't merged into master for four years now, despite it being in a working condition and a couple of people expressing interest. Great way to encourage contribution. 

I need it for work and have to port it forward 700+ commits now. Oh joy ...

eric-wieser, 2019-11-25:
> and have to port it forward 700+ commits now.

I'd be surprised if you have to change much at all, other than perhaps makefiles for the example project. Does it merge cleanly?

Fatsie, 2019-11-25:
I am working on it and plan to upstream it.
Is it urgent ? I have been busy with other cleanup patches on which Wishbone will be rebased.
See https://gitlab.com/Chips4Makers/cocotb/tree/dev_wishbone

mkreider, 2019-11-25:
I was planning to merge master with the old code tonight and try to get my test cases underway ASAP,
but I'll do that for use within the company only. So take your time

m-kru, 2020-04-23:
@Fatsie what is the status?