cocotb/cocotb #869: Mechanism for waiting for the first of multiple triggers doesn't work for Joins

eric-wieser, 2019-03-17:
Consider
```python
@cocotb.coroutinue
def task_one():
    yield Timer(10)

@cocotb.coroutinue
def task_two():
   yield Timer(2)
```

It's not possible to work out which one finishes first:
```python
@cocotb.coroutinue
def test_with_fork():
   co_one = cocotb.fork(task_one)
   co_one = cocotb.fork(task_two)
   co_which = yield [co_one, co_two]
   assert co_which in {co_one, co_two}  # fails, co_which  is None
```

`asyncio.wait` solves this by returning `Future` objects, rather than the result of the futures.

garmin-mjames, 2019-09-26:
We could add a `Future` trigger that has access to both the `Outcome` and `RunningCoroutine` objects.

Alternatively we could add a trigger that returns just the `RunningCoroutine` object, an alternate to `Join` like `Ident` or similar, so the test would explicitly yield on a trigger expecting the coroutine back.

```
co_which = yield [Ident(co_one), Ident(co_two)]
```

ktbarrett, 2024-04-26:
I think  we are going to leave `First` and `Combine` the same and introduce something newer that works better. xref #2659. xref #3583.

ktbarrett, 2024-06-25:
You can use the following as of #3931.

```python3
trigger = await First(task1.join(), task2.join())
first_task = trigger.task
# work out which one
if trigger is task1.join():
    ...
```

closed 2025-01-08 by ktbarrett

marlonjames, 2025-01-09:
As of #4341, use this pattern:

```py3
trigger = await First(task1.complete, task2.complete)
first_task = trigger.task
# work out which one
if first_task is task1:
    ...
```

PR #4341 (2025-01-08): Refactor `Join` into `TaskComplete`
Closes #869.

Since `Join(task)` and `task.join()` behave the same as `task` when using the resulting value, these syntaxes were updated to simply return the Task. This allows the Join trigger to work like all other triggers by returning itself when awaited.

The Join trigger  was renamed to TaskComplete and is now avaiable via the `task.complete` attribute.

Storing the TaskComplete trigger on the Task means we don't need to use ParameterizedSingletonMetaclass to associate their lifetimes and preserve uniqueness.